Joshua-Riek/ubuntu-rockchip #911: sources.list.d package missing on upgraded installs

SimplyCorbett, 2024-07-07:
I just upgraded some old 24.04 Orange Pi 5 plus installs and then installed a fresh version on a new unit. I noticed in /etc/apt/sources.list.d the juries-Ubuntu-panfork-mesa-noble.sources exists on the new unit, but it doesn’t exist on the older installs that were upgraded.

I think they were version 2.1.0 installs before the upgrade.

Do I just add this repository or are there other changes that I need to do for the upgrade?

Joshua-Riek, 2024-07-07:
I have no idea what "juries-Ubuntu-panfork-mesa-noble.sources" is, the juries prefix seems like you modified something.

SimplyCorbett, 2024-07-07:
> I have no idea what "juries-Ubuntu-panfork-mesa-noble.sources" is, the juries prefix seems like you modified something.

Oh for the — I hate my spell corrector. I’m going to look into how to disable my spell corrector on this device, sorry.

It’s jjriek-Ubuntu-panfork-mesa-noble.sources, but it’s missing on the upgraded installs.

SimplyCorbett, 2024-07-08:
I’m just reinstalling the servers with the latest version but its weird that wasn’t included in the upgrade.

Joshua-Riek, 2024-07-08:
It should not be a problem, there where a lot of changes to how i built the rootfs after the initial Ubuntu 24.04 release to accommodate for building Ubuntu 22.04 and 24.04 with the same script.

closed 2024-07-08 by Joshua-Riek