Joshua-Riek/ubuntu-rockchip #913: Request support LubanCat RK3588

ambitiousCC, 2024-07-08:
Hi. This project already support LubanCat 4.

As the newest addition to the Luban Cat series, Luban Cat 5 supports the RK3588 chip, which theoretically provides better performance than the RK3588S. However, when I use the LubanCat 4 image on the LubanCat 5, I still encounter issues like a green screen. I hope the developer can make this version of the mainboard compatible.

Joshua-Riek, 2024-07-08:
Ive not been able to find anything about the Luban Cat 5, do you have a link to the device?

ambitiousCC, 2024-07-12:
Sorry, I've been quite busy these past few days and haven't replied promptly. The official documentation can be found at: https://doc.embedfire.com/products/link/en/latest/linux/ebf_lubancat_doc.html

ambitiousCC, 2024-07-12:
> 我一直找不到有关 Luban Cat 5 的任何信息，您有该设备的链接吗？

Sorry, I've been quite busy these past few days and haven't replied promptly. The official documentation can be found at: https://doc.embedfire.com/products/link/en/latest/linux/ebf_lubancat_doc.html

ambitiousCC, 2024-07-12:
What specific information do you need about the Lubancat 5? The official English documentation doesn't seem to be updated. I found the Chinese version: https://doc.embedfire.com/products/link/zh/latest/linux/ebf_lubancat.html, which includes hardware information for the Lubancat 5.

ben-auo, 2024-11-29:
https://doc.embedfire.com/linux/rk3588/quick_start/zh/latest/quick_start/lubancat/lubancat5.html

https://www.amazon.com/TP-Motherboard-Source-Linux-Development-0GB/dp/B0D8L17D8H

closed 2025-02-10 by github-actions[bot]