Joshua-Riek/ubuntu-rockchip #914: Rootfs opensource for LubanCat4 RK3588s

shingoxy, 2024-07-08:
Hi, anywhere I could find the open souce of ubuntu rootfs for LubanCat4? 20.04 version will be fine.

Thanks~

Joshua-Riek, 2024-07-08:
Ubuntu 20.04 is not supported.

closed 2024-07-08 by Joshua-Riek

shingoxy, 2024-07-09:
Thanks @Joshua-Riek , if I changed to your version 22.04 or 24.04, could I get the rootfs directly? Thanks~

shingoxy, 2024-07-09:
> The rootfs can be downloaded from all GitHub action builds:
> 
> * https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/nightly.yml
> 
> Direct download links:
> 
> * [ubuntu-22.04-preinstalled-desktop-arm64-rootfs](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9831289481/artifacts/1675870374)
> * [ubuntu-22.04-preinstalled-server-arm64-rootfs](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9831289481/artifacts/1675826002)
> * [ubuntu-24.04-preinstalled-desktop-arm64-rootfs](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9831289481/artifacts/1676046419)
> * [ubuntu-24.04-preinstalled-server-arm64-rootfs](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9831289481/artifacts/1675927144)

Thanks~That's really helpful, and where could I get the kernel. I am doing some develop with new cameras.

Joshua-Riek, 2024-07-09:
Kernel source code is below, on branch noble and jammy.
https://github.com/Joshua-Riek/linux-rockchip