Joshua-Riek/ubuntu-rockchip #919: Kernel 1019 broken on at least 2 devices

lukaszsobala, 2024-07-12:
Hello,

I've been using your distribution for a while and it's been quite stable. But now I encountered a serious issue.

On Orange Pi 3B and Radxa Rock 5B, after updating to kernel 1019 (6.1), most things stopped working, both in the GUI and the CLI. Error: "failed to fork", then (often) "Segmentation fault" (core dumped). If it was 1 device I'd think that it's just a bad eMMC or something, but it happens on 2 independently so it must be something with the system.

Joshua-Riek, 2024-07-12:
Similar issue to #906, i really need help with this problem. There are over 6000 new kernel commits..

lukaszsobala, 2024-07-12:
Could you tell me how to revert to 1017? I can't even change the vmlinuz link (this is enough, right?). I'll try to do it externally on the memory card but Opi3b is booting from eMMC which I'd prefer not to disassemble.

Thanks

Joshua-Riek, 2024-07-12:
I have to head to work right now and will not be able to provide exact steps at the moment. But in U-Boot you can select the kernel if you connect uart cable.

lukaszsobala, 2024-07-12:
Thank you, will try this

Joshua-Riek, 2024-07-12:
@lukaszsobala if you can could you try https://github.com/Joshua-Riek/ubuntu-rockchip/issues/906#issuecomment-2211842830 instead? If this works i can push an update with this as a hot fix. Even though this is a hack It's better than the alternative right now.

Joshua-Riek, 2024-07-12:
Ubuntu 24.04 server information for the Rock 5B using the Rockchip Linux 6.1 rkr1 SDK. 
> Kernel versions 6.1.0-1017-rockchip and below use the rkr1 SDK.

## Kernel
```bash
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 6.1.0-1016-rockchip #16-Ubuntu SMP Thu Jun  6 00:11:54 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
```
## Memory 
```bash
ubuntu@ubuntu:~$ cat /proc/meminfo
MemTotal:       16345992 kB
MemFree:        15407088 kB
MemAvailable:   15837728 kB
Buffers:           29932 kB
Cached:           515132 kB
SwapCached:            0 kB
Active:           235244 kB
Inactive:         355612 kB
Active(anon):       2120 kB
Inactive(anon):    61608 kB
Active(file):     233124 kB
Inactive(file):   294004 kB
Unevictable:       34324 kB
Mlocked:           26224 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 4 kB
Writeback:             0 kB
AnonPages:         80196 kB
Mapped:            77888 kB
Shmem:             10416 kB
KReclaimable:      66148 kB
Slab:             206620 kB
SReclaimable:      66148 kB
SUnreclaim:       140472 kB
KernelStack:        4800 kB
PageTables:         2316 kB
SecPageTables:         0 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     8172996 kB
Committed_AS:     484228 kB
VmallocTotal:   133143592960 kB
VmallocUsed:       14888 kB
VmallocChunk:          0 kB
Percpu:             3136 kB
CmaTotal:         262144 kB
CmaAllocated:      42724 kB
CmaReleased:      219420 kB
CmaFree:          219420 kB
HugePages_Total:       0
HugePages_Free:        0
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:               0 kB
ubuntu@ubuntu:~$ free -mh 
               total        used        free      shared  buff/cache   available
Mem:            15Gi       496Mi        14Gi        10Mi       596Mi        15Gi
Swap:             0B          0B          0B
```
## Limits, running processes, open files
```bash
ubuntu@ubuntu:~$ ulimit -a
real-time non-blocking time  (microseconds, -R) unlimited
core file size              (blocks, -c) 0
data seg size               (kbytes, -d) unlimited
scheduling priority                 (-e) 0
[... 235 lines cut ...]
root        2860       2  0 14:17 ?        00:00:00 [kworker/5:1-inode_switch_wbs]
root        2861       2  0 14:17 ?        00:00:00 [kworker/5:3-inode_switch_wbs]
root        2862       2  0 14:17 ?        00:00:00 [kworker/5:4-inode_switch_wbs]
root        2863       2  0 14:17 ?        00:00:00 [kworker/5:5-inode_switch_wbs]
root        2864       2  0 14:17 ?        00:00:00 [kworker/5:6-inode_switch_wbs]
root        2865       2  0 14:17 ?        00:00:00 [kworker/5:7-inode_switch_wbs]
root        2866       2  0 14:17 ?        00:00:00 [kworker/5:8-inode_switch_wbs]
root        2867       2  0 14:17 ?        00:00:00 [kworker/5:9-inode_switch_wbs]
root        2868       2  0 14:17 ?        00:00:00 [kworker/5:10-inode_switch_wbs]
root        2869       2  0 14:17 ?        00:00:00 [kworker/5:11-inode_switch_wbs]
root        2870       2  0 14:17 ?        00:00:00 [kworker/5:12-inode_switch_wbs]
root        2871       2  0 14:17 ?        00:00:00 [kworker/5:13-events]
root        2872       2  0 14:17 ?        00:00:00 [kworker/5:14-inode_switch_wbs]
root        2873       2  0 14:17 ?        00:00:00 [kworker/5:15]
root        2874       2  0 14:17 ?        00:00:00 [kworker/2:0-events]
root        2875       2  0 14:17 ?        00:00:00 [kworker/2:3]
root        2876       2  0 14:17 ?        00:00:00 [kworker/7:2-events]
root        2879       2  0 14:17 ?        00:00:00 [kworker/1:0H]
root        3001       2  0 14:17 ?        00:00:00 [psimon]
root        3247       2  0 14:17 ?        00:00:00 [kworker/7:3-events]
root        3248       2  0 14:17 ?        00:00:00 [kworker/7:0H]
ubuntu      3249    2162 99 14:17 pts/0    00:00:00 ps -ef
ubuntu@ubuntu:~$ lsof | wc -l
1267
```

lukaszsobala, 2024-07-12:
I'm sorry for mistyping, I actually didn't notice this happening on Rock 5B (16 GB) but on Rock 5A (4 GB RAM)!

The Opi3b has 8 GB of RAM (it is the older version, not the one with refreshed WiFi) and the issue occurs.

lukaszsobala, 2024-07-12:
@Joshua-Riek I managed to apply the workaround but it did not fix the problem (Opi 3b). Here are my limits:
```
real-time non-blocking time  (microseconds, -R) unlimited
core file size              (blocks, -c) 0
data seg size               (kbytes, -d) unlimited
scheduling priority                 (-e) 0
file size                   (blocks, -f) unlimited
pending signals                     (-i) 500000
max locked memory           (kbytes, -l) 983568
max memory size             (kbytes, -m) unlimited
open files                          (-n) 1024
pipe size                (512 bytes, -p) 8
POSIX message queues         (bytes, -q) 819200
real-time priority                  (-r) 0
stack size                  (kbytes, -s) 8192
cpu time                   (seconds, -t) unlimited
max user processes                  (-u) 500000
virtual memory              (kbytes, -v) unlimited
file locks                          (-x) unlimited
```

Is it possible this is the open files limit?

Joshua-Riek, 2024-07-12:
Open files limit could also need to be increased.

lukaszsobala, 2024-07-13:
I tried the other workaround too, on both boards, it is still broken.

lukaszsobala, 2024-07-13:
Setting TasxMax set to infinity in `/usr/lib/systemd/system/user-.slice.d/10-defaults.conf`seems to have an effect on Rock 5A.

Joshua-Riek, 2024-07-13:
I found the issue after some painful bisecting:
- https://github.com/Joshua-Riek/linux-rockchip/commit/9b0de6bdec732c6b4dea6c82c76414fe81b172df
- https://github.com/Joshua-Riek/linux-rockchip/commit/c91ccee082cee846d78100f4efc40aef9041352b

The fix has been tagged for Ubuntu-rockchip-6.1.0-1020.20 and will be pushed out later today.

closed 2024-07-13 by Joshua-Riek

lukaszsobala, 2024-07-13:
Thank you very much @Joshua-Riek. On Opi3b I am having trouble executing anything, is there a way to somehow install the new kernel despite this? I rarely can run even `sudoedit` and `apt update && apt upgrade` seems insurmountable...

Joshua-Riek, 2024-07-13:
You can select the kernel to boot by connecting a serial cable, there will be a menu in uboot.

lukaszsobala, 2024-07-13:
What if the only ones I have now are both broken? 1017 and 1019 (although I need to confirm 1017 is broken as well).

Joshua-Riek, 2024-07-13:
1017 is not broken

lukaszsobala, 2024-07-13:
Indeed, it works OK now after booting from serial and removing 1019. Thanks again

Joshua-Riek, 2024-07-13:
I'll have 1020 uploaded later tonight.

cantalupo555, 2024-07-14:
> Olá,
> 
> Eu uso sua distribuição há algum tempo e ela está bastante estável. Mas agora encontrei um problema sério.
> 
> No Orange Pi 3B e Radxa Rock 5A, após a atualização para o kernel 1019 (6.1), a maioria das coisas parou de funcionar, tanto na GUI quanto na CLI. Erro: "falha ao bifurcar", então (frequentemente) "Falha de segmentação" (núcleo despejado). Se fosse um dispositivo, eu pensaria que é apenas um eMMC ruim ou algo assim, mas acontece em 2 independentemente, então deve ser algo com o sistema.

They've been having this problem for a week now.
I had to reinstall the system and not run `sudo apt dist-upgrade`.

cantalupo555, 2024-07-14:
> Hello,
> 
> I've been using your distribution for a while and it's been quite stable. But now I encountered a serious issue.
> 
> On Orange Pi 3B and Radxa Rock 5A, after updating to kernel 1019 (6.1), most things stopped working, both in the GUI and the CLI. Error: "failed to fork", then (often) "Segmentation fault" (core dumped). If it was 1 device I'd think that it's just a bad eMMC or something, but it happens on 2 independently so it must be something with the system.

I've been having this problem for almost 2 weeks.
I had to reinstall the system and not run `sudo apt dist-upgrade`.

lukaszsobala, 2024-07-14:
@cantalupo555 this is fixed now