Joshua-Riek/ubuntu-rockchip #921: Support for Hardkernel Odroid M1 (RK3568B2)

Dr3nz4r, 2024-07-13:
Hi there, first of all awesome project youre doing the gods work 🙏

is there an option to make a 22.04/24.04 for the odroid m1 with hardware video and npu support for tools like jellyfin/plex/immich ?

I'm happy to give you a sponsor as well for your efforts

Joshua-Riek, 2024-07-16:
I really need put a disclaimer when it comes to asking me to support new hardware, there have been too many requests.

I work on boards that I find interesting, and when I do find interesting hardware I reach out to the company and hope they can send me a sample as I cant afford all these boards. People don't understand the development time that is required when adding support for new hardware it's quite a large commitment on my part. Additionally I do not get paid for my work, this entire project is just a hobby. Because this project is getting so large and there is a lot of hardware that I do not have a significant interest in, I will require someone to fund the development process, this will not be cheap.

But I'm happy to accept any pull request if someone else wants to add support for new hardware.

Dr3nz4r, 2024-07-16:
Totally get you. Just noticed that you have other rk35xx boards in there already and thought it might be just a few kernel options away 👀 sorry but I'm a total noob who bought an sbc from a company that doesn't maintain any of the software

Joshua-Riek, 2024-07-16:
Normally, I can port a board in a few hours. But it depends on the hardware components used and how well the drivers work as sometimes there can be incompatibilities, I've found time and time again that companies are using new hardware components that do not have good driver or software support and this can be very time consuming.

Dr3nz4r, 2024-07-16:
If you need anything, i have it running and can dump you anything you need or make pictures of the pcb if that helps

koutheir, 2024-09-04:
I'm also interested in getting this board working. Are there any instructions on how to build the Ubuntu images from sources? If I can do that, then I'll create a pull request with the needed changes to resolve this issue.

koutheir, 2024-09-04:
FYI, I flashed `ubuntu-24.04-preinstalled-desktop-arm64-roc-rk3588s-pc.img.xz` (release v2.3.1) just to try it out. Booting failed with the following serial console output:

```plain
...
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
1063 bytes read in 20 ms (51.8 KiB/s)
U-Boot menu
1:	Ubuntu 24.04 LTS 6.1.0-1023-rockchip
2:	Ubuntu 24.04 LTS 6.1.0-1023-rockchip (rescue target)
Enter choice: 1:	Ubuntu 24.04 LTS 6.1.0-1023-rockchip
Retrieving file: /boot/initrd.img-6.1.0-1023-rockchip
22304079 bytes read in 1878 ms (11.3 MiB/s)
Retrieving file: /boot/vmlinuz-6.1.0-1023-rockchip
53680640 bytes read in 4496 ms (11.4 MiB/s)
append: root=UUID=348e79d2-40ba-4792-ae34-947783823bac rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory quiet splash plymouth.ignore-serial-consoles
Retrieving file: /lib/firmware/6.1.0-1023-rockchip/device-tree/rockchip-odroid-m1.dtb
** File not found /lib/firmware/6.1.0-1023-rockchip/device-tree/rockchip-odroid-m1.dtb **
Skipping l0 for failure retrieving fdt
2:	Ubuntu 24.04 LTS 6.1.0-1023-rockchip (rescue target)
Retrieving file: /boot/initrd.img-6.1.0-1023-rockchip
22304079 bytes read in 1878 ms (11.3 MiB/s)
Retrieving file: /boot/vmlinuz-6.1.0-1023-rockchip
53680640 bytes read in 4497 ms (11.4 MiB/s)
append: root=UUID=348e79d2-40ba-4792-ae34-947783823bac rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory splash plymouth.ignore-serial-consoles single
Retrieving file: /lib/firmware/6.1.0-1023-rockchip/device-tree/rockchip-odroid-m1.dtb
** File not found /lib/firmware/6.1.0-1023-rockchip/device-tree/rockchip-odroid-m1.dtb **
Skipping l0r for failure retrieving fdt
SCRIPT FAILED: continuing...
...
```

koutheir, 2024-09-06:
@Joshua-Riek, is there any documentation or pointers on how to do this porting?

Joshua-Riek, 2024-09-06:
Ubuntu has little to no documentation for building the operating system. Most of everything I have done and know has been because of some form of reverse engineering over a long period of time.

So my only pointer is to dump hours of your life into the code and figure it out.

koutheir, 2024-09-06:
I see. Thank you for your efforts.