Joshua-Riek/ubuntu-rockchip #927: Orange Pi 5 Max

lotharmatheos, 2024-07-16:
Hello,
Which image should I use for new Orange Pi 5 Max?

5 Plus? Or 5 Pro?

Joshua-Riek, 2024-07-16:
Is this device out yet?

lotharmatheos, 2024-07-16:
> Is this device out yet?

Yes,
I bought [one](https://www.aliexpress.us/item/1005007348974947.html?spm=a2g0o.productlist.main.1.4b3ekUPmkUPmhx&algo_pvid=b61143f0-8c47-401f-9c66-3d25309dd7b2&algo_exp_id=b61143f0-8c47-401f-9c66-3d25309dd7b2-0&pdp_npi=4%40dis%21PLN%21616.18%21616.18%21%21%21153.75%21153.75%21%402103850917211444769064977efcc7%2112000040367586785%21sea%21PL%21136799574%21&curPageLogUid=aHkguFhwwmlI&utparam-url=scene%3Asearch%7Cquery_from%3A) on ali.
Official page:
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-Max.html

lotharmatheos, 2024-07-17:
> Wondering the same. Is there a build for the new Orange Pi 5 Max? Specially interested in using the pcie3x4 for an nvme drive

Orange Pi 5 Plus has the same m.2 slot and 2.5Gb LAN
But Wifi 6E is only on Max. Different DRAM is not a problem, I think

Joshua-Riek, 2024-07-17:
I'll reach out to Orange Pi to get a development kit, but I do expect the WiFi / Bluetooth to not work due to a driver conflict between the SDIO and PCIe versions of the BCMDHD driver.

DiegoBM, 2024-07-17:
Actually if that were the case it would still work out in my case since I won't be needing those, but I presume it's quite important to most.
I hope you don't mind me asking, but how long does it take usually for one of these new boards to get fully supported? (giving for granted that it depends on how much time it can be put into it and that it will be the roughest of guesses)

Joshua-Riek, 2024-07-17:
Should be ready a few days after I get the hardware - if there are no other hardware or driver problems besides the previously mentioned WiFi / Bluetooth issues. I'll likely release an image with both WiFi and Bluetooth **fully disabled** until a proper solution can be found.

DiegoBM, 2024-07-17:
That would be incredibly useful! Thank you @Joshua-Riek for your efforts!

xutianq999, 2024-07-23:
I have tried to use the image of 5 plus/5pro on orange pi 5max and failed. I used RKdevtool to burn the image into emmc, but I used the official burning configuration of orangepi. I don’t know if there is a problem in this aspect, and finally it caused the screen to be stuck after booting.
[image]
[image]

Joshua-Riek, 2024-07-23:
I just got the Orange Pi 5 Max and CM5 in the mail, I will work on them tonight.

xutianq999, 2024-07-23:
Thank you so much

Joshua-Riek, 2024-07-24:
I added initial support for both devices in U-Boot and the kernel. I just need to do some testing to make sure everything is working. 

I did see that the Orange Pi 5 Max uses a different wireless chip than the Orange Pi 5 Pro, so maybe I can get it working correctly.

DiegoBM, 2024-07-24:
Thank you for the update @Joshua-Riek! Hopefully it will all go smoothly!

Joshua-Riek, 2024-07-25:
Below are logs from my attempt to get WiFi working via a DKMS package i made for BCMDHD. Sadly it does not work yet.
```
[   16.078083] dhd_static_buf_sdio: loading out-of-tree module taints kernel.
[   16.090900] [dhd] STATIC-MSG) dhd_static_buf_init : 101.10.361.36 (wlan=r892223-20231107-1)
[   16.091042] [dhd] STATIC-MSG) dhd_init_wlan_mem : prealloc ok for index 0: 8459264(8261K)
[   16.173869] [dhd] _dhd_module_init: in Dongle Host Driver, version 101.10.591.52.27 (20240409-1)(20240411-2)(d83d8d7)
               /var/lib/dkms/bcmdhd-sdio/101.10.591.52.27-1/build/src compiled on Jul 24 2024 at 22:08:26
[   16.173968] [dhd] dhd_wlan_init_gpio: WL_HOST_WAKE=-1, oob_irq=150, oob_irq_flags=0x4
[   16.173971] [dhd] dhd_wlan_init_gpio: WL_REG_ON=-1
[   16.173973] [dhd] dhd_wifi_platform_load: Enter
[   16.173975] [dhd] Power-up adapter 'DHD generic adapter'
[   18.092934] [dhd] dummy_probe: func->num=0x1; 
[   18.092946] [dhd] dummy_probe: class=0xff; vendor=0x6cb; device=0xaabf
[   18.093110] [dhd] dummy_probe: func->num=0x2; 
[   18.093113] [dhd] dummy_probe: class=0xff; vendor=0x6cb; device=0xaabf
[   18.093145] [dhd] wifi_platform_set_power = 1, delay: 200 msec
[   18.093148] [dhd] ======== PULL WL_REG_ON(-1) HIGH! ========
[   18.406172] [dhd] wifi_platform_set_power = 1, sleep done: 200 msec
[   18.406186] [dhd] wifi_platform_bus_enumerate device present 1
[   18.406189] [dhd] ======== Card detection to detect SDIO card! ========
[   18.422331] [dhd] bcmsdh_register: register client driver
[   18.422515] [dhd] bcmsdh_sdmmc_probe: Enter num=1
[   18.422636] [dhd] bcmsdh_sdmmc_probe: Enter num=2
[   18.422639] [dhd] bus num (host idx)=2, slot num (rca)=1, caps=0x4008010d
[   18.422642] [dhd] found adapter info 'DHD generic adapter'
[   18.422648] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 3, size 139264
[   18.422677] [dhd] succeed to alloc static buf
[   18.422680] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 4, size 0
[   18.422766] [dhd] sdioh_attach: set sd_f2_blocksize 256
[   18.422921] [dhd] sdioh_attach: sd clock rate = 150000000
[   18.423196] [dhd] dhdsdio_probe : no mutex held
[   18.423203] [dhd] dhdsdio_probe : set mutex lock
[   18.423267] [dhd] F1 signature read @0x18000000=0x1040aabf
[   18.425611] [dhd] F1 signature OK, socitype:0x1 chip:0xaabf rev:0x0 pkg:0x4
[   18.426056] [dhd] DHD: dongle ram size is set to 1310720(orig 1310720) at 0x170000
[   18.426098] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 7, size 43088
[   18.426134] [dhd] dhd_conf_set_chiprev : devid=0xaabf, chip=0xaabf, chiprev=0
[   18.426245] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 0, size 10320
[   18.426304] [dhd] dhd_ioctl_entry_local invalid parameter net 0000000000000000 dev_priv 00000000b9700a34
[   18.426315] [dhd] CFG80211-ERROR) wl_is_fils_supported : FILS NOT supported, err -22
[   18.426815] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 5, size 65536
[   18.427321] [dhd] CFG80211-ERROR) wl_is_fils_supported : FILS NOT supported, err -19
[   18.427514] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 19, size 65712
[   18.427750] [dhd] dhd_attach(): thread:dhd_watchdog_thread:371 started
[   18.427883] [dhd] dhd_attach(): thread:dhd_dpc:372 started
[   18.427970] [dhd] dhd_dpc_thread: set dhd_dpc_prio 98
[   18.428076] [dhd] dhd_attach(): thread:dhd_rxf:373 started
[   18.428092] [dhd] dhd_deferred_work_init: work queue initialized
[   18.428100] [dhd] dhd_tcpack_suppress_set: TCP ACK Suppress mode 0 -> mode 2
[   18.428106] [dhd] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=40, TCPDATA_INFO_MAXNUM=40
[   18.428176] [dhd] sdioh_cis_read: func_cis_ptr[0]=0x10ac
[   18.432089] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 1, size 10300
[   18.432109] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 2, size 65536
[   18.432144] [dhd] dhdsdio_probe_init: making DHD_BUS_DOWN
[   18.432229] [dhd] Dongle Host Driver, version 101.10.591.52.27 (20240409-1)(20240411-2)(d83d8d7)
               /var/lib/dkms/bcmdhd-sdio/101.10.591.52.27-1/build/src compiled on Jul 24 2024 at 22:08:26
[   18.432895] [dhd] Register interface [wlan0]  MAC: c0:f5:35:bb:67:99
[   18.433030] [dhd] [wlan0] wl_android_wifi_off :  g_wifi_on=1 force_off=1
[   18.433041] [dhd] dhd_bus_devreset: == Power OFF ==
[   18.433160] [dhd] dhd_bus_stop: making DHD_BUS_DOWN
[   18.433184] [dhd] bcmsdh_oob_intr_unregister: Enter
[   18.433190] [dhd] bcmsdh_oob_intr_unregister: irq is not registered
[   18.433201] [dhd] dhd_bus_devreset: making dhdpub up FALSE
[   18.433207] [dhd] dhd_txglom_enable: enable 0
[   18.433214] [dhd] dhd_bus_devreset: making DHD_BUS_DOWN
[   18.433239] [dhd] wifi_platform_set_power = 0, delay: 10 msec
[   18.433247] [dhd] ======== PULL WL_REG_ON(-1) LOW! ========
[   18.550579] [dhd] wifi_platform_set_power = 0, sleep done: 10 msec
[   18.550593] [dhd] [wlan0] wl_android_wifi_off : out
[   18.550608] [dhd] wl_android_post_init: 0
[... 53 lines cut ...]
[   21.623897] [dhd] dhd_txglom_enable: enable 1
[   21.623907] [dhd] dhd_conf_set_txglom_params : txglom_mode=copy
[   21.623914] [dhd] dhd_conf_set_txglom_params : txglomsize=36, deferred_tx_len=0
[   21.623920] [dhd] dhd_conf_set_txglom_params : txinrx_thres=128, dhd_txminmax=-1
[   21.623926] [dhd] dhd_conf_set_txglom_params : tx_max_offset=0, txctl_tmo_fix=300
[   21.623936] [dhd] dhd_conf_get_disable_proptx : fw_proptx=1, disable_proptx=-1
[   21.625984] [dhd] dhd_wlfc_hostreorder_init(): successful bdcv2 tlv signaling, 64
[... cut ...]
[   21.727872] [dhd] CFG80211-ERROR) __wl_update_wiphybands : error reading txbf_bfe_cap (-23)
[   21.728390] [dhd] CFG80211-ERROR) __wl_update_wiphybands : error reading txbf_bfr_cap (-23)
[   21.737212] [dhd] [wlan0] wl_cfg80211_up : Roam channel cache enabled
[   21.738807] [dhd] [wlan0] dhd_open : Exit ret=0
[   21.738824] [dhd] [wlan0] dhd_pri_open : tx queue started
[   21.738830] [dhd] dhd_pri_open : mutex is released.
[   21.964511] [dhd] P2P interface registered
[   21.982414] [dhd] P2P interface started
```

xutianq999, 2024-07-25:
That's a pity, wouldn't it be better to communicate with their official staff? Anyway, thank you for your work.

Joshua-Riek, 2024-07-25:
I'd probably be told to use a different kernel configuration, but that would require me to maintain a new kernel just for the 5 Max and 5 Pro. This is not something I don't want to do. I'll still tinker when I can. Seeing the firmware load correctly is a good start.

Joshua-Riek, 2024-07-25:
I made a few tweaks to the DKMS package and WiFi is now working.
 
[image]

xutianq999, 2024-07-26:
wow，That looks great!

DiegoBM, 2024-07-26:
Hard to believe this happens so fast! Thank you!
Roughtly, when do you usually roll out new releases?

lotharmatheos, 2024-07-26:
Amazing!

Thanks!

Joshua-Riek, 2024-07-29:
I just finished all the kernel changes for the release. I need to do some testing then package the dkms for WiFi, hopefully I will have the release out this week.

djholt, 2024-08-02:
What great timing for mine to arrive in the mail. :)

Thanks for your quick work on this!! @Joshua-Riek

Joshua-Riek, 2024-08-02:
I'm building a release now. Hopefully I can publish it tonight after some quick smoke testing.

djholt, 2024-08-02:
I'll test with nvme as soon as you push it.

closed 2024-08-02 by Joshua-Riek

djholt, 2024-08-02:
@Joshua-Riek It's beautiful. Thanks for being awesome!

DiegoBM, 2024-08-03:
Thank you for completing this so quickly @Joshua-Riek!
Apologies for the silly question but once I install one of your releases, how do I continue getting updates? Will the standard apt commands get the changes from your new releases from that point onwards? Thanks!

watou, 2024-08-03:
> Will the standard apt commands get the changes from your new releases from that point onwards?

@DiegoBM yes!

Joshua-Riek, 2024-08-03:
All updates can be obtained from apt. There is no need to re-install every release.

lotharmatheos, 2024-08-12:
Tested and it's working ;-)
Minor glitches at graphics rendering, but Ethernet, Wifi, NVMe work flawlessly.
Sent my gratitude $ ;-)

Below NVME performance comparison beetween Orange Pi 5 Max and Raspberry Pi
[image]

DiegoBM, 2024-08-12:
@lotharmatheos what application are you running for those IO tests?

I also tested the distro yesterday on mi 5 Max, and it seemed to work like a charm! I didn't do a thorough test (network and stuff) but instead, like @lotharmatheos I spent some time testing multiple NVME cards to check performance, only in my case I did install KDiskMark on the desktop version, and many devices were in the 3K+ MB/s! Now that's proper usable storage!

lotharmatheos, 2024-08-12:
> @lotharmatheos what application are you running for those IO tests?
> 
> I also tested the distro yesterday on mi 5 Max, and it seemed to work like a charm! I didn't do a thorough test (network and stuff) but instead, like @lotharmatheos I spent some time testing NVME performance, only in my case it was mutliple different cards and I did install KDiskMark on the desktop version, and many devices were in the 3K+ MB/s! Now that's proper usable storage!

Used open source tools (hdparm, dd, FIO ald IOZone) in form of script

https://github.com/TheRemote/PiBenchmarks

Joshua-Riek, 2024-08-13:
I've never seen RK3588 at 3GHz. Maybe it's possible with LN2 and some hardware hacking.

DummyTitan, 2024-09-01:
Have any of you been able to make the Orange Pi 5 Max Hardware accelerator work?

Joshua-Riek, 2024-09-01:
What's a hardware accelerator?

DummyTitan, 2024-09-02:
i meant the vpu, the drivers work but the dts (device tree module) doesn't have &av1d that is required for transcoding av1. I'm wondering Joshua-Riek might you know how to fix that cause every time I try to fix it and compile it just gives me a syntax error and I'm not exactly a pro at this so i might be missing something.

DiegoBM, 2024-09-02:
Is anyone else experiencing issues connecting to WiFi? I can connect to the 2.4G network allright, but nmcli will refuse connecting to the 5G network with the "Secrets were required, but not provided" error

Joshua-Riek, 2024-09-02:
> i meant the vpu, the drivers work but the dts (device tree module) doesn't have &av1d that is required for transcoding av1. I'm wondering @Joshua-Riek might you know how to fix that cause every time [...]

You can open a proper issue and / or create a pull request, I have very little time right now for development.

DiegoBM, 2024-09-02:
Thank you, [done ](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1026) as suggested

denis1482, 2024-09-11:
Since Orange Pi 5 Max WiFi problem is fixed, can it help solving Orange Pi 5 Pro WiFi problem?

jimmyhon, 2024-09-13:
> I'll reach out to Orange Pi to get a development kit, but I do expect the WiFi / Bluetooth to not work due to a driver conflict between the SDIO and PCIe versions of the BCMDHD driver.

@Joshua-Riek Did you happen to also get the schematics for the board? Their Google Drive is empty. I'm trying to upstream the device tree into mainline.

https://github.com/jimmyhon/linux/pull/1