Joshua-Riek/ubuntu-rockchip #930: Support for AIO-3588L

serwansj, 2024-07-17:
Hi,

I would like to use your images on the AIO-3588L (https://en.t-firefly.com/product/industry/aio3588l). Would it be possible to support this board? If necessary I can post the device tree or any other information.

Thanks!

Joshua-Riek, 2024-07-17:
I'm happy to review a pull request..

serwansj, 2024-07-19:
so i added the dts files for the device tree to the kernel in https://github.com/Joshua-Riek/linux-rockchip, however, i used the u-boot from the `roc-rk3588s-pc` board. the config i used is:

```
# shellcheck shell=bash

export BOARD_NAME="AIO 3588L"
export BOARD_MAKER="Firefly"
export BOARD_SOC="Rockchip RK3588"
export BOARD_CPU="ARM Cortex A76 / A55"
export UBOOT_PACKAGE="u-boot-radxa-rk3588"
export UBOOT_RULES_TARGET="roc-rk3588s-pc-rk3588s"

function config_image_hook__aio-3588l() {
    local rootfs="$1"

    # Install panfork
    chroot "${rootfs}" add-apt-repository -y ppa:jjriek/panfork-mesa
    chroot "${rootfs}" apt-get update
    chroot "${rootfs}" apt-get -y install mali-g610-firmware
    chroot "${rootfs}" apt-get -y dist-upgrade

    # Install libmali blobs alongside panfork
    chroot "${rootfs}" apt-get -y install libmali-g610-x11

    # Install the rockchip camera engine
    chroot "${rootfs}" apt-get -y install camera-engine-rkaiq-rk3588

    return 0
}
```
and then i also hardcoded the name of the u-boot in the build script since it was trying to find one called `aio-3588l*`.

managed to build the image successfully, however it does not boot and i get `rk3x-i2c timeout` errors. i assume i also need to change something for u-boot, but i am not really sure what, could you maybe point me in the right direction?

Joshua-Riek, 2024-07-19:
Your system is likely not loading the right device tree since. The device tree to load during boot is defined in U-Boot. A specific U-Boot config will need to be made for this device. 

Or you could chroot and define the device tree manually.
https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-LTS#changing-the-device-tree

serwansj, 2024-07-20:
I think i managed to configure U-Boot correctly since I am not getting the previous error anymore and I/O is now being detected. However, now I encounter the same issue as in https://github.com/Joshua-Riek/ubuntu-rockchip/issues/149. Reading through that issue I  do not understand how to transfer the fix to my problem.

Can you give me a tiphon this one as well? I assume it should be the last hurdle.

serwansj, 2024-07-20:
figured it out, thanks for the help!

will make a PR in the coming days :)

closed 2024-08-02 by Joshua-Riek