Joshua-Riek/ubuntu-rockchip #941: virgl_test_server works fine in Ubuntu 22.04, but not in Ubuntu 24.04?

khanumballz, 2024-07-24:
I normally use this for my hardware acceleration for a smoother Blender 2.79 experience on Distrobox/BoxBuddy:

```
khan@khan-desktop:~$ virgl_test_server
MESA-LOADER: failed to open rknpu: /usr/lib/dri/rknpu_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
vtest_client_dispatch_commands: client context created.
client: VTEST_CLIENT_DISCONNECTED
```

and then in my Ubuntu 18 box:

```
khan@ubuntu-18-box:~$ blender
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip
*** stack smashing detected ***: <unknown> terminated
Aborted (core dumped)
khan@ubuntu-18-box:~$ 
```

That's the output that I get. Like I said, it works fine in Ubuntu 22.04 but not in 24.04 for some reason, either Wayland or Xorg doesn't matter.

This is the only thread I've found which reports anything remotely similar to this bug:

https://github.com/Joshua-Riek/ubuntu-rockchip/issues/527

I will keep investigating and in the meantime simply switch SD card if I ever need hardware acceleration in Blender.

Cheers

martinbone, 2024-08-16:
@khanumballz  You may be interested that I've found an alternative solution that avoids this issue. I've been able to run Blender v2.78 (x86_64) on Joshua's Ubuntu 24.04 via box64 (latest version). And it automatically runs with hardware acceleration. There are some issues (especially when trying to render). But for model building it's working fine. I downloaded the version of Blender from the official website, and just run the Blender executable file from it's folder.

khanumballz, 2024-08-17:
> @khanumballz You may be interested that I've found an alternative solution that avoids this issue. I've been able to run Blender v2.78c (x86_64) on Joshua's Ubuntu 24.04 via box64 (latest version). And it automatically runs with hardware acceleration. There are some issues (especially when trying to render). But for model building it's working fine. I downloaded the version of Blender from the official website, and just run the Blender executable file from it's folder.

Thanks. 

I've decided to settle for using Ubuntu 22.04 for the time being, seen as I have other pieces of software that I need to be able to run smoothly, such as OBS studio and Ultimaker Cura.

I suppose it is a question of how much one values one's time, but if the Orange Pi 5 Plus had better software and driver support, it would be an unimaginably powerful device, and easily a low-budget desktop replacement capable of running on just 20w of power (which is particularly attractive for running Large Language Models, computer vision, image generation, etc.)

martinbone, 2024-09-07:
> @khanumballz You may be interested that I've found an alternative solution that avoids this issue. I've been able to run Blender v2.78c (x86_64) on Joshua's Ubuntu 24.04 via box64 (latest version). [...]

For anyone else who's interested - I'm actually now using Bforartists (x86_64) version 0.9.5 (based on Blender 2.79b), which works better.

khanumballz, 2024-09-08:
Are you also running it using box64?

martinbone, 2024-09-08:
> Are you also running it using box64?

Yes ...

GuEe-GUI, 2024-09-20:
Hello, Did you fix it?

RadxaYuntian, 2024-09-20:
This is not something we supported. In addition, doesn't virgl needs to be run within VM? If your system can detect RKNPU then it is likely not running in a VM.

GuEe-GUI, 2024-09-20:
> This is not something we supported. In addition, doesn't virgl needs to be run within VM? If your system can detect RKNPU then it is likely not running in a VM.

virgl is not only work for a VM or QEMU, I have tried on `forlinx 3588` again, it's OK.

closed 2024-12-16 by github-actions[bot]