Joshua-Riek/ubuntu-rockchip #944: uefi ubuntu version

Baddad23, 2024-07-25:
Looking for a version of ubuntu that can be installed to nvme using orangepi-5plus_UEFI_Release_v0.11.2.img? I don't know if you have or are working on it. I presently have win11 2h24 installed on 1tb nvme and looking to install ubuntu on second nvme.

Joshua-Riek, 2024-07-25:
UEFI Ubuntu would be an installation medium that boots from a device like a USB. This introduces a lot of problems for the platform.

If you want to put in the development time, I would be happy to review a pull request. But I have absolutely no intent of producing a UEFI image myself.

closed 2024-07-25 by Joshua-Riek

Baddad23, 2024-07-25:
I guess I'm not explaining myself very well, sorry. When installing windows for amd64 on an Opi5+ you first have to use RKDevtool to write both:

Uefi = orangepi-5plus_UEFI_Release_v0.11.2.img
Loader = rk3588_spl_loader_v1.15.113.bin

the latest files and this won't change unless there are newer releases put out. And then start the opi5+ with what is essentially a standard bootable Windows install usb drive which will install a copy of win11 on a hard drive or in my case a nvme ssd. The uefi can recognize different drives and even change boot order of these drives.  So I'm looking for the same thing for a ubuntu install. Just like you would install unbuntu on a PC. Create a bootable ubuntu install usb and use it to install opi5+ ubuntu to a nvme ssd just like I would if I were doing it on a PC. You would no longer need RKDevtool to install either windows or ubuntu. You create a boot install usb drive using Rufus or Etcher or any other utility and install latest OS.

I hope I'm explaining it correctly.

Joshua-Riek, 2024-07-25:
I fully understand what you are talking about, but i'm trying to tell you its not feasible. Just look at how Ubuntu handles the Raspberry Pi, there are no UEFI Ubuntu installers made for the Raspberry Pi. 

While I'm sure that I could make it work with a lot of hacks, I'm not going to spend months of my free time to develop something that I do not want to.

Baddad23, 2024-07-25:
Ok, sorry to have bothered you.