Joshua-Riek/ubuntu-rockchip #947: KVM does not work in Ubuntu 24.04

FastestLion9038, 2024-07-26:
Running command "kvm-ok" on Orange Pi 5 Plus reports "KVM Acceleration can be used" but trying to do anything related to KVM does not work.

Joshua-Riek, 2024-07-26:
Please provide exact reproducible steps and errors encountered.

FastestLion9038, 2024-07-26:
I followed this tutorial [https://jobcespedes.dev/2023/11/running-virtual-machines-on-orange-pi-5/](https://jobcespedes.dev/2023/11/running-virtual-machines-on-orange-pi-5/) and used command:

virt-install -n test --memory 1024 --arch aarch64 --vcpus 1 \
  --disk /var/lib/libvirt/images/test.qcow2,device=disk,bus=virtio \
  --os-variant=cirros0.5.2 \
  --nographic \
  --boot loader=/usr/share/AAVMF/AAVMF_CODE.fd,loader.readonly=yes,loader.type=pflash

Using Virt-Manager had some success but it would never boot and it had some issues about the "Relaxed State" for the cpu XML configuration.

I also used [https://github.com/dockur/windows-arm/](https://github.com/dockur/windows-arm/) but it failed at the point when it would try to boot the Windows VM due to KVM issues.

Running Command "kvm-ok" says KVM can be used but in reality, any VM using KVM will not boot. Only pur QEMU VM's will actually boot.

FastestLion9038, 2024-07-26:
Update

Turned out to be a user permission problem for accessing /dev/kvm. Adding myself to the kvm group fixed things slightly.
Now I have an error reporting: "kvm: Cannot enable KVM when guest CPU has EL3 enabled"

FastestLion9038, 2024-07-26:
Nevermind, Fixed all my issues by using virt-manager directly after adding myself to the /dev/kvm group. Got past that EL3 error by using virt-manager instead of command line to start the VM.

Thanks @Joshua-Riek

closed 2024-07-26 by FastestLion9038