Joshua-Riek/ubuntu-rockchip #949: Ubuntu 22.04 not completing its boot up

amarmashruwala, 2024-07-27:
Hi there. i was working with docker on my orange pi 5 plus running Ubuntu 2204 and i am not sure if i did something wrong. Now when i turn on the orange pi 5 plus, it stuck on the boot image and does not complete the boot. moving circle. Is there a way i can fix this without needing to reinstall Ubuntu on my installed NVME? thank you.

amarmashruwala, 2024-07-27:
Hi Guys, i get the following error when irun this command
Sudo dd if=/lib/u-boot-orangepi-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc
dd: failed to open '/lib/u-boot-orangepi-rk3588/rkspi_loader.img': No such file or directory

Any idea why? thanks

wmport, 2024-07-27:
Logs are needed, show the output of the command
dmesg | grep ppower
 dmesg | grep drm

I have errors in the nanopc-t6 power logs.... but it shuts down normally.
 dmesg | grep power
[    8.908247] thermal_sys: Registered thermal governor 'power_allocator'
[    9.582327] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up pcie-supply from device tree
[    9.582349] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up pcie-supply property in node /power-management@fd8d8000/power-controller failed
[    9.601764] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up vdpu-supply from device tree
[    9.601782] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up vdpu-supply property in node /power-management@fd8d8000/power-controller failed
[    9.601823] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rkvdec0-supply from device tree
[    9.601838] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rkvdec0-supply property in node /power-management@fd8d8000/power-controller failed
[    9.601963] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up vcodec-supply from device tree
[    9.601979] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up vcodec-supply property in node /power-management@fd8d8000/power-controller failed
[    9.608803] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up audio-supply from device tree
[    9.608828] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up audio-supply property in node /power-management@fd8d8000/power-controller failed
[    9.620363] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga30-supply from device tree
[    9.620387] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga30-supply property in node /power-management@fd8d8000/power-controller failed
[    9.620786] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga31-supply from device tree
[    9.620807] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga31-supply property in node /power-management@fd8d8000/power-controller failed
[    9.623582] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up venc0-supply from device tree
[    9.623604] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up venc0-supply property in node /power-management@fd8d8000/power-controller failed
[    9.624048] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up venc1-supply from device tree
[    9.624069] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up venc1-supply property in node /power-management@fd8d8000/power-controller failed
[    9.625005] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rkvdec1-supply from device tree
[    9.625026] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rkvdec1-supply property in node /power-management@fd8d8000/power-controller failed
[    9.625485] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up av1-supply from device tree
[    9.625506] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up av1-supply property in node /power-management@fd8d8000/power-controller failed
[    9.999732] rockchip-vop2 fdd90000.vop: [drm:vop2_wait_power_domain_off] *ERROR* wait pd0 off timeout power_ctrl: 0x0
[   10.091547] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gpu-supply from device tree
[   10.091577] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gpu-supply property in node /power-management@fd8d8000/power-controller failed
[   10.097050] W : [File] : drivers/gpu/arm/bifrost/platform/rk/mali_kbase_config_rk.c; [Line] : 144; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.
[   10.126419] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up sdio-supply from device tree
[   10.126439] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up sdio-supply property in node /power-management@fd8d8000/power-controller failed
[   10.126487] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gmac-supply from device tree
[   10.126496] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gmac-supply property in node /power-management@fd8d8000/power-controller failed
[   10.126527] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu-supply from device tree
[   10.126536] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu-supply property in node /power-management@fd8d8000/power-controller failed
[   10.130572] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up nputop-supply from device tree
[   10.130603] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up nputop-supply property in node /power-management@fd8d8000/power-controller failed
[   10.130671] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu1-supply from device tree
[   10.130697] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu1-supply property in node /power-management@fd8d8000/power-controller failed
[   10.130739] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu2-supply from device tree
[   10.130765] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu2-supply property in node /power-management@fd8d8000/power-controller failed
[   39.716254] rockchip-vop2 fdd90000.vop: [drm:vop2_power_domain_off_by_disabled_vp] *ERROR* unexpected power on pd6
[   39.716277] rockchip-vop2 fdd90000.vop: [drm:vop2_power_domain_off_by_disabled_vp] *ERROR* unexpected power on pd5

Joshua-Riek, 2024-07-27:
> Hi Guys, i get the following error when irun this command Sudo dd if=/lib/u-boot-orangepi-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc dd: failed to open '/lib/u-boot-orangepi-rk3588/rkspi_loader.img': No such file or directory
> 
> Any idea why? thanks

That path does not exist, please review the wiki link below.
https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-LTS#install-u-boot-to-an-sd-card-or-emmc

As for the reason why your system does not boot there is no way for me to tell why unless you provide uart logs.

Joshua-Riek, 2024-07-27:
@wmport if there is an issue with the nanopc t6 please open a new issue.