Joshua-Riek/ubuntu-rockchip #950: OrangePi 3B won't boot from NVMe drive

lbnascimento, 2024-07-27:
Using the latest release (v2.2.1) of Ubuntu 22.04, the OrangePi 3B won't boot from an NVMe drive (stuck at the OPi logo). The same image works perfectly when flashed to an SD card.

Both devices were flashed using balenaEtcher, the SD card was flashed on a USB card reader and the NVMe drive was flashed on a USB-M.2 adapter.

Gi-0n, 2024-07-27:
I'm not an expert but I also have the same board, I would try another method, you can follow the steps in the
[Wiki - 22.04 LTS](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-LTS)

If it doesn't work for you, try version 24.04. My opi3b board (v2.1) works better with Ubuntu 24.04.

dmitrii-kuzmin, 2024-07-27:
Do you have sd card inserted? Have you tried to boot when the sd slot is empty?

a-nanasov, 2024-07-27:
Do you have sd card inserted? Have you tried to boot when the sd slot is empty?

defencedog, 2024-07-30:
> I'm not an expert but I also have the same board, I would try another method, you can follow the steps in the [Wiki - 22.04 [...]

I also have opi3b board (v2.1) ... you said 24.04 is working better for you. Can you explain why? Wireless, HDMI audio / video, Bluetooth are working properly? Have you tried `KODI` & does it give GPU / VPU acceleration? Have you tried HW accelerated `JellyFin` in docker container?

I know it's lot to ask but you can really help by providing a solid comprehensive feedback

lbnascimento, 2024-07-30:
> I'm not an expert but I also have the same board, I would try another method, you can follow the steps in the [Wiki - 22.04 [...]

I tried 24.04 and it also didn't boot. I also tried `ubuntu-rockchip-install` and it didn't work, both with 22.04 and 24.04.

One thing I noticed from an image that does work (the official Ubuntu image from the OPi website) is that it has an EFI partition, while this one doesn't. Does OPi 3B only support EFI boot from NVMe drives?

Gi-0n, 2024-07-30:
> I also have opi3b board (v2.1) ... you said 24.04 is working better for you. Can you explain why? Wireless, HDMI audio / video, Bluetooth are working properly? Have you tried `KODI` & does it give [...]

22.04 was not stable for me, I had continuous crashes 

HDMI works for me
AUDIO only works for me through HDMI through 3.5mm JACK, there is no sound
GPU works in 3D applications like Quakespasm, Freedoom, Flare, DevilutionX (768P from 57fps to 60fps)
the only negative is that I have graphical errors (icons disappearing) due to the current version of Panfrost
WIFI: It works and gives me a minimum speed of 10mbps (I read that another user whose router was next to him only gave 5mbps)
BLUETOOH: Cannot find device
VPU: Not tested yet
DOCKER-KODI-JELLYFIN: I have not tried it, I mostly use UBUNTU 24.04 as a desktop

defencedog, 2024-08-01:
@lbnascimento were you able to boot from NVMÈ?because I couldn't... I used `RKDEV` tool & **USB-A <> USB-A** cable

lbnascimento, 2024-09-04:
> @lbnascimento were you able to boot from NVMÈ?because I couldn't... I used `RKDEV` tool & **USB-A <> USB-A** cable

I didn't, and my NVMe also died in the middle of my tests, so I decided to just buy an SD card to make my life easier.

czf0613, 2024-10-21:
I met the same problem. My board version is v2.1.
Everything works fine on SD card, but it won't boot after I burn just the same image to nvme SSD.
I'm sure that I correctly setup the SPI flash following the official instruction, the official image can successfully boot from nvme SSD.

Sincerely waiting for solution. Thanks!

defencedog, 2024-10-22:
@czf0613 I have no problem booting from NVME SBC v2.1
helpful links:
https://github.com/defencedog/orangepi3b_v2.1/blob/main/Clone_TFcard_NVME.md
https://github.com/defencedog/orangepi3b_v2.1/tree/main/Armbian_Noble_rk6.1.75

czf0613, 2024-10-22:
> @czf0613 I have no problem booting from NVME SBC v2.1 helpful links: https://github.com/defencedog/orangepi3b_v2.1/blob/main/Clone_TFcard_NVME.md https://github.com/defencedog/orangepi3b_v2.1/tree/main/Armbian_Noble_rk6.1.75

Still not work. I don't know what went wrong because I could successfully boot OrangePi's official image on nvme.
I've tried other boards like Radxa 5ITX, OrangePi 5Pro, 5Plus and another 3B board. Emmm, it looks a little bit strange. Some of them works but most of them not, here are my tests:

1, Rock 5ITX and OrangePi 5Pro can successfully boot from nvme drive. But Rock 5 ITX often fails to reboot. (I don't know why and it only occurred several times after the first boot, but now it works fine.)

2, OrangePi 5Plus and two OrangePi 3B all failed to boot from nvme. I am sure that I have already updated the SPI flash because all of them can load the OrangePi's official image from nvme.

defencedog, 2024-10-23:
I think you have flashed your SPI boot. Do this
1] Unplug NVME
2] Install Armbian .img from my repo on sdcard
3] Boot from it

Report back if it is booting

If yes re-flash your SPI flash memory to use Armbian `uboot` https://github.com/defencedog/orangepi3b_v2.1/blob/main/Clone_TFcard_NVME.md

Zibi108, 2024-10-23:
Hei, i'm glad i've found this forum! 
I've got opi3b for more than a year now. Don't know what version, can't dismount the case just to find out... 
I have successfully installed different images on NVME, a dozen times or more. And i have several SD cards with different editions, mostly different flavors of Armbian. 

I really would like to be able to install Josh Riek's ubuntu image on the internal SSD. It works fine from SD. It is unexcelled in two counts: 
Video playback! It's so far the only one that can play Youtube at 1080p. I've got only 3% frame loss (at 1080p/25fps), which -- for this hardware -- is *pretty* watchable. Not all other images play so smooth at 720p. 
Secondly, i am partial to Cinnamon desktop, and this edition has the newest and most beautiful version of it. (All ubu noble armbian images have Cinnamon half broken.) And wi-fi works just fine, which is not so obvious for other distros. 

For two days i tried to run it from nvme and quite positively, or rather negatively i should say, and no go. I tried each-and-every-thing you mention above, and more. 
Today, while having my morning coffee, i flashed the nvme with another image which i never tried, and it boots just like that (snap!)
I still have the Josh'es image  on SD card and i'm open for experiments. 

Just an idea: once i updated SPI from some of the newest (june this year) armbian images, and none of the older images would boot.  Including the one on internal. The newest one worked, though. Erased the SPI, flashed again from the older opi factory image, everything works fine again. 
So maybe that is important, after all, from what image SPI is flashed?

Zibi108, 2024-10-23:
Hei @czf0613  
> 
> After some trying, I noticed something that might be the key to the issue. You can try these steps to boot. 1, At first I burn the Ubuntu Image in this project to SD card, and boot from SD card. 2, Then, use the `u-boot-install-mtd` command to flash the SPI flash. 3, Then you can sucessfully boot from nvme drive.
> 

So, after flashing SPI from this system, do other -- older! & other distros -- SD cards work as well? Have you tried?

czf0613, 2024-10-23:
Distros on SD card won't be affected.
I have not tried other distros on nvme drive, you can have a try :)

Zibi108, 2024-10-23:
Wow, it works!  
So, from the ubuntu-24.04-preinstalled system running from SD card, I flashed the image onto internal nvme using Image Writer (gnome-disk-utility), then flashed SPI with the u-boot-install-mtd command. *) Yep, it boots up! 

After the initial setup i shut down the system, only to check if the older SD cards still work. Yep, all fine. 
Now i have some work to do, to configure the system on internal nvme -- it will stay there for a while. 
Thanks a lot! 
_________
*) Sure, i know that you know all of this -- i just write for anybody who may have the same problem and google this thread in a future.

closed 2024-12-30 by github-actions[bot]