Joshua-Riek/ubuntu-rockchip #951: Failed power in logs  nanopc-t6 24.04 latest builds

wmport, 2024-07-27:
I have errors in the nanopc-t6 power logs.... but it shuts down normally.
latest builds

dmesg | grep power
[ 8.908247] thermal_sys: Registered thermal governor 'power_allocator'
[ 9.582327] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up pcie-supply from device tree
[ 9.582349] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up pcie-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.601764] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up vdpu-supply from device tree
[ 9.601782] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up vdpu-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.601823] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rkvdec0-supply from device tree
[ 9.601838] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rkvdec0-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.601963] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up vcodec-supply from device tree
[ 9.601979] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up vcodec-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.608803] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up audio-supply from device tree
[ 9.608828] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up audio-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.620363] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga30-supply from device tree
[ 9.620387] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga30-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.620786] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga31-supply from device tree
[ 9.620807] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga31-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.623582] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up venc0-supply from device tree
[ 9.623604] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up venc0-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.624048] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up venc1-supply from device tree
[ 9.624069] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up venc1-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.625005] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rkvdec1-supply from device tree
[ 9.625026] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rkvdec1-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.625485] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up av1-supply from device tree
[ 9.625506] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up av1-supply property in node /power-management@fd8d8000/power-controller failed
[ 9.999732] rockchip-vop2 fdd90000.vop: [drm:vop2_wait_power_domain_off] ERROR wait pd0 off timeout power_ctrl: 0x0
[ 10.091547] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gpu-supply from device tree
[ 10.091577] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gpu-supply property in node /power-management@fd8d8000/power-controller failed
**[ 10.097050] W : [File] : drivers/gpu/arm/bifrost/platform/rk/mali_kbase_config_rk.c; [Line] : 144; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.**
[ 10.126419] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up sdio-supply from device tree
[ 10.126439] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up sdio-supply property in node /power-management@fd8d8000/power-controller failed
[ 10.126487] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gmac-supply from device tree
[ 10.126496] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gmac-supply property in node /power-management@fd8d8000/power-controller failed
[ 10.126527] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu-supply from device tree
[ 10.126536] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu-supply property in node /power-management@fd8d8000/power-controller failed
[ 10.130572] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up nputop-supply from device tree
[ 10.130603] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up nputop-supply property in node /power-management@fd8d8000/power-controller failed
[ 10.130671] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu1-supply from device tree
[ 10.130697] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu1-supply property in node /power-management@fd8d8000/power-controller failed
[ 10.130739] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu2-supply from device tree
[ 10.130765] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu2-supply property in node /power-management@fd8d8000/power-controller failed
**[ 39.716254] rockchip-vop2 fdd90000.vop: [drm:vop2_power_domain_off_by_disabled_vp] ERROR unexpected power on pd6
[ 39.716277] rockchip-vop2 fdd90000.vop: [drm:vop2_power_domain_off_by_disabled_vp] ERROR unexpected power on pd5**

Joshua-Riek, 2024-07-27:
Is there an actual problem here?

wmport, 2024-07-28:
When switching iptv channels to mpv the player is started from the command line, playback of the stream is periodically intentionally stopped.
The only way to close mpv is to force close the terminal window.
This error appears in the logs

JS ERROR: TypeError: this.actor is null
_syncEnabled@resource:///org/gnome/shell/ui/windowManager.js:145:25
onStopped@resource:///org/gnome/shell/ui/windowManager.js:157:35
_makeEaseCallback/<@resource:///org/gnome/shell/ui/environment.js:65:22
_easeActorProperty/<@resource:///org/gnome/shell/ui/environment.js:232:60
_destroyWindowDone@resource:///org/gnome/shell/ui/windowManager.js:1607:21
onStopped@resource:///org/gnome/shell/ui/windowManager.js:1575:39
_makeEaseCallback/<@resource:///org/gnome/shell/ui/environment.js:65:22
_easeActor/<@resource:///org/gnome/shell/ui/environment.js:154:64
@resource:///org/gnome/shell/ui/init.js:21:20

Maybe they're connected

[ 10.097050] W : [File] : drivers/gpu/arm/bifrost/platform/rk/mali_kbase_config_rk.c; [Line] : 144; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.

[ 39.716254] rockchip-vop2 fdd90000.vop: [drm:vop2_power_domain_off_by_disabled_vp] ERROR unexpected power on pd6
[ 39.716277] rockchip-vop2 fdd90000.vop: [drm:vop2_power_domain_off_by_disabled_vp] ERROR unexpected power on pd5

Tiemichael, 2024-07-28:
I can confirm this behaviour (NanoPC-T6, Ubuntu 24.04 installed 4 days ago)
No idea if this causes any problems