Joshua-Riek/ubuntu-rockchip #952: Orange Pi 5 Plus Freezing / Crashing with HDMI Out

hoeken, 2024-07-27:
I installed Ubuntu 24.04 on my OrangePi 5 Plus back in April, and recently pulled it back out today.  I did an `apt-get update` and `apt-get upgrade` and now it crashes when a monitor is connected via HDMI.  It seems to run fine when nothing is connected.  Sometimes it will freeze with an unresponsive desktop, sometimes it will flash the desktop for a few seconds and then go into some sort of boot loop.  I'm not sure if it makes any difference, but I am booting off a NVME.

Some debug:

```
boatpi@orangepi:~$ uname -a
Linux orangepi 6.1.0-1020-rockchip #20-Ubuntu SMP Sat Jul 13 19:58:54 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
```

dmesg from after plugging in HDMI:
```
[  956.473123] dwhdmi-rockchip fdea0000.hdmi: i2c read err!
[  956.473174] dwhdmi-rockchip fdea0000.hdmi: get hdcp1.4 capable failed:-5
[  956.575528] dwhdmi-rockchip fdea0000.hdmi: i2c read time out!
[  956.575567] dwhdmi-rockchip fdea0000.hdmi: get hdcp1.4 capable failed:-11
[  956.613059] Console: switching to colour frame buffer device 480x135
[  956.613141] rockchip-drm display-subsystem: [drm] fb0: rockchipdrmfb frame buffer device
[  957.090165] dwhdmi-rockchip fdea0000.hdmi: use tmds mode
[  957.090228] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 3840x2160p60, type: 11(if:1000, flag:0x0) for vp1 dclk: 594000000
[  957.090779] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] dclk_out1 div: 0 dclk_core1 div: 2
[  957.090816] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx_ropll_cmn_config bus_width:5aa320 rate:5940000
[  957.091111] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx phy pll locked!
[  957.091133] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] set dclk_vop1 to 594000000, get 594000000
[  957.091209] dwhdmi-rockchip fdea0000.hdmi: final tmdsclk = 594000000
[  957.202362] dwhdmi-rockchip fdea0000.hdmi: don't use dsc mode
[  957.202380] dwhdmi-rockchip fdea0000.hdmi: dw hdmi qp use tmds mode
[  957.202391] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: bus_width:0x5aa320,bit_rate:5940000
[  957.202602] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx phy lane locked!
[  957.339673] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp1
[  957.358042] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx_ropll_cmn_config bus_width:8 rate:5940000
[  957.358313] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx phy pll locked!
[  958.148427] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 3840x2160p60, type: 11(if:1000, flag:0x0) for vp1 dclk: 594000000
[  958.148591] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx_ropll_cmn_config bus_width:5aa320 rate:5940000
[  958.148859] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx phy pll locked!
[  958.148912] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] dclk_out1 div: 0 dclk_core1 div: 2
[  958.148926] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] set dclk_vop1 to 594000000, get 594000000
[  958.148978] dwhdmi-rockchip fdea0000.hdmi: final tmdsclk = 594000000
[  958.258866] dwhdmi-rockchip fdea0000.hdmi: don't use dsc mode
[  958.258874] dwhdmi-rockchip fdea0000.hdmi: dw hdmi qp use tmds mode
[  958.258880] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: bus_width:0x5aa320,bit_rate:5940000
[  958.259071] rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx phy lane locked!
```

Joshua-Riek, 2024-07-27:
There where a lot of HDMI changes in the latest kernel sdk. I know very little about the vop2 driver, but what resolution is your monitor?

hoeken, 2024-07-27:
It's a 4k monitor

thanhtantran, 2024-08-02:
I have same problem. After upgrade the HDMI out and desktop is crashed, but connect via SSH still working.

this is my dmesg

[dmesg.txt](https://github.com/user-attachments/files/16468471/dmesg.txt)

Joshua-Riek, 2024-08-02:
Would be good to raise this issue with Rockchip on redmine, but I don't have any contacts at Rockchip or access to their issue tracker.

ewaldc, 2024-10-01:
I also have a (cluster of) Orange Pi 5+'s with Ubuntu 24.04 and several 4K monitors that are also booting from NVME, but have not seen this issue.  I sometimes boot them from SD card or EMMC memory, no HDMI problem there as well.  One system is even "hacked" to run Linux Mint 22 Cinnamon, totally unsupported, but HDMI is fine.   
I have the same (strange) HDMI dmesg output (including all error messages like `lane locked` etc.) as posted above, so if there is an issue it is not captured by these messages.  It is possible that multiple HDMI drivers are attempting to claim the HDMI interface (hence the errors), but I do not know for sure...
The only issue I have seen is that the HDMI signal is not being turned off after a long time of inactivity, which on one of my monitors is causing issues (have to power it physically off for like 20 seconds, and then power back on).

The one thing that is definately different: I made my first Ubuntu 24.04 install a bit later in May.  So maybe something in the early stages of the 24.04 upgrade processes has not gone well.
What if you install a fresh, recent copy of the OS e.g. on SD card: does HDMI crash as well then?

closed 2025-02-10 by github-actions[bot]