Joshua-Riek/ubuntu-rockchip #956: Bug Report: HDMI signal not sent after TV is restarted

jackduckworth, 2024-07-28:
### What happened?

I have an OPi5 plugged into my TV via HDMI. I'm running Ubuntu 24.04.

Everything works fine until I turn the TV off and on again. The screen is now black and I cant get the display to come back on.

Similarly, if I turn the OPi5 on before the TV then I dont get any signal.

In both cases I can still connect via ssh, and everything seems healthy. But the only way, that I can find, to get the display back is to do a reboot.

### Kernel version

6.1.0-1020-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
Dim Screen:    off
Screen Blank:  Never
```

Joshua-Riek, 2024-07-28:
When this problem occurs, can you login though ssh and post the output of `dmesg`?

GutsyHeart, 2024-07-29:
Similar problem with a monitor connected via Display Port
[dmesg.log](https://github.com/user-attachments/files/16409630/dmesg.log)

jackduckworth, 2024-07-30:
I seem to get this when I turn off the TV

> rtw_8821cu 5-1:1.2: failed to download firmware
> rtw_8821cu 5-1:1.2: leave idle state failed
> rtw_8821cu 5-1:1.2: failed to send h2c command
> rtw_8821cu 5-1:1.2: failed to send h2c command
> rtw_8821cu 5-1:1.2: failed to leave ips state
> rtw_8821cu 5-1:1.2: failed to leave idle state
> 

and this when I turn it on again

> rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp0
> rockchip-vop2 fdd90000.vop: [drm:vop2_power_domain_off_by_disabled_vp] *ERROR* unexpected power on pd6
> rockchip-vop2 fdd90000.vop: [drm:vop2_power_domain_off_by_disabled_vp] *ERROR* unexpected power on pd5

[dmesg2.txt](https://github.com/user-attachments/files/16427123/dmesg2.txt)

jackduckworth, 2024-08-05:
I was poking around with the cables behind the TV and just noticed this seems to be working now? 

I noticed some kernel updates last time I did an update but wasn't expecting this to be fixed already.

And these message are no longer visible in dmesg:
> rtw_8821cu 5-1:1.2: failed to download firmware
> rtw_8821cu 5-1:1.2: leave idle state failed
> rtw_8821cu 5-1:1.2: failed to send h2c command
> rtw_8821cu 5-1:1.2: failed to send h2c command
> rtw_8821cu 5-1:1.2: failed to leave ips state
> rtw_8821cu 5-1:1.2: failed to leave idle state
>

closed 2025-02-10 by github-actions[bot]