Joshua-Riek/ubuntu-rockchip #962: widevine not supported

nrpetonr, 2024-07-31:
### What happened?

websites that require widevine are not working (amazon music, disney+, etc.) .i tried browser spoofing, did not work anymore.

### Kernel version

6.1.0-1021-rockchip

### SBC model

Orange PI 5+

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Joshua-Riek, 2024-07-31:
Widevine has never worked on Ubuntu 24.04. Only on Ubuntu 22.04.

nrpetonr, 2024-08-02:
> Widevine has never worked on Ubuntu 24.04. Only on Ubuntu 22.04.

was not aware, thx. i just isntalled latest Ubuntu 22.04 release 2.3.0 just to test it is still working. ufnortunately not, even with spoofing. is it supposed to be still working?

Joshua-Riek, 2024-08-02:
It's not supposed to work on Ubuntu 24.04. But it should work on Ubuntu 22.04.

Also, there is no need to re-install your system when there is a new release. All updates and changes are available through apt.

nrpetonr, 2024-08-02:
> It's not supposed to work on Ubuntu 24.04. But it should work on Ubuntu 22.04.
> 
> Also, there is no need to re-install your system when there is a new release. All updates and changes are available through apt.

i was on 24.04 ;-) 
any special instructions to make it work on 22.04 or it should work out of the box?

nrpetonr, 2024-09-01:
use this [script ](https://github.com/AsahiLinux/widevine-installer) followed by [browser spoofing,](https://chromewebstore.google.com/detail/user-agent-switcher-and-m/bhchdcejhohfmigjafbampogmaanbfkg?hl=en) to fix the problem with drm sources. enjoy...

closed 2024-09-01 by nrpetonr