Joshua-Riek/ubuntu-rockchip #963: Feature Request: support RK3528A

lukaszsobala, 2024-07-31:
### Feature Description

Hello,

Some new boards are coming out soon with the entry-level Rockchip RK3528A chip and up to 4 GB of RAM. Would it be worth supporting these or are 4xA53 cores too slow?

Thanks

Joshua-Riek, 2024-08-01:
Adding support for an entire SOC is impossible, it needs to be done by a board by board basis. 

I will be transparent, I'm starting to care less and less about new hardware. Primarily because hardware vendors basically dump poorly made hardware on me while I do all the work for free and support a better operating system than what they can provide. I have insider accounts from employees of multiple companies admitting this internally and how they no longer care about supporting hardware themselves, they can just dump everything on me. I will likely start to reject any and all new hardware unless someone creates a methodical pull request that follows strict code quality standards, I find personal interest in a device, or I am paid to support / maintain the device. 

I will never ask for money from users, though its greatly appreciated. However I'm exhausted from being used by so many companies, I now need to be an asshole to protect myself, which it's sad that it's gotten to this point.

SimplyCorbett, 2024-08-01:
I’m of the opinion that you shouldn’t be adding so much new hardware unless other people join in and help out. You’re already supporting a -lot- of hardware. Don’t overwork yourself and burn out. The project should be fun, not a job/chore.

I mean it’s a bit different if they pay you but they aren’t currently paying you and you have no obligations to support their hardware.

I use the orange pi 5 plus (multiple) and I haven’t seen any effort from that developer to change how their stuff works. The files are still hosted on google drive (including the distros) which are inaccessible most of the time because of download limits. They are essentially forcing users to use third party options.

I’m not even sure if they update any of their distros.

my two cents.

dfloer, 2024-08-03:
> I will be transparent, I'm starting to care less and less about new hardware. Primarily because hardware vendors basically dump poorly made hardware on me while I do all the work for free and support a better operating system than what they can provide. I have insider information from multiple companies admitting this internally and how they no longer care about supporting hardware themselves, they can just dump everything on me. I will likely start to reject any and all new hardware unless someone creates a methodical pull request that follows strict code quality standards, I find personal interest in a device, or I am paid to support / maintain the device.
> 
> I will never ask for money from users, though its greatly appreciated. However I'm exhausted from being used by so many companies, I now need to be an asshole to protect myself, which it's sad that it's gotten to this point.

Trying to get basic info out of some of these board vendors to try to make device tree overlays to enable hardware they advertise in their spec pages has definitely opened my eyes up to this. It seems certain vendors are just crapping out new boards without thinking about any sort of support.

What I'm trying to say, is I feel your pain, and think your response is entirely reasonable. These companies can't even be bothered to support their own hardware.

lukaszsobala, 2024-08-03:
That's totally fair, no problem. Maybe I'll try to hack it together by substituting the device tree for example. Hopefully it can work, maybe I'll try to format & submit this as you explained. 

I was asking precisely because vendor OSes are so bad, they use repositories that often go down, or just have limited driver support. With vanilla Ubuntu the repo issues are not likely at least... It's as if those who make the distro don't actually use it/have it running to test. But I guess this is all known here. Issue can be closed I guess?

SimplyCorbett, 2024-08-07:
> That's totally fair, no problem. Maybe I'll try to hack it together by substituting the device tree for example. Hopefully it can work, I will attempt to format & submit this as you explained.
> 
> I was asking precisely because vendor OSes are so bad, they use repositories that often go down, or just have limited driver support. With vanilla Ubuntu the repo issues are not likely at least... It's as if those who make the distro don't actually use it/have it running to test. But I guess this is all known here. Issue can be closed I guess?

I mean... I personally wouldn't touch most of their software unless it's from a US vendor. I've heard there are a lot of questionable things on i.e. the orange pi 5 stuff. (just heard, I don't know for sure. Never verified).

I'm just confused on why they aren't paying Joshua considering he is basically the sole reason these boards have anyone purchasing them still. I'm not discounting Armbian, but Joshua puts in a ton of work to make these boards functional. If Joshua were to leave I'm not sure if the Armbian team would be able to take over or not leaving most of us stranded.

I've given serious thought to using intel nuc units instead because of it *shrug*.

lukaszsobala, 2024-08-07:
@SimplyCorbett I guess hearsay is quite an effective way of spreading propaganda. Just because something comes from China doesn't mean it's filled with spyware.

SimplyCorbett, 2024-08-07:
> @SimplyCorbett I guess hearsay is quite an effective way of spreading propaganda. Just because something comes from China doesn't mean it's filled with spyware.

Well I would have tested to verify but I simply CANNOT download their operating systems. The google mirror always says download limit reached.

and I don’t care enough to check several times a day for the limit to go away.

closed 2024-08-10 by Joshua-Riek