Joshua-Riek/ubuntu-rockchip #966: Signal messenger database error

fyptog, 2024-08-01:
### What happened?

For the last few days Signal messenger gives a "Database error", and needs to be reset, restarted and re-linked every time the app is started. There are 4 orange pi5 computers being used as personal desktops in my household (standalone solar house) and all are having the same problem. very frustrating. 
No sign of Signal problems on our PCs with MX linux, or PopOS (snap).

### Kernel version

6.1.0-1021-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Joshua-Riek, 2024-08-01:
It may be worth opening an issue with the Signals application, though i did find https://github.com/signalapp/Signal-Desktop/issues/5721, can you try starting the program with `--disable-gpu`?

fyptog, 2024-08-01:
Still the same Database Error.
Thanks anyway.

fyptog, 2024-08-03:
This is it - https://github.com/snapcrafters/signal-desktop/issues/300

Joshua-Riek, 2024-08-03:
Closing as this is not an issue with Ubuntu Rockchip.

closed 2024-08-03 by Joshua-Riek