Joshua-Riek/ubuntu-rockchip #967: In 24.04 LTS, I want to enable these features by default and deploy Redroid using Docker.

124ffee, 2024-08-04:
### Feature Description

CONFIG_PSI=y
CONFIG_IKHEADERS=y
CONFIG_KPROBES=y
CONFIG_MAC80211_HWSIM=y
CONFIG_ASHMEM=y
CONFIG_ANDROID_BINDER_IPC=y
CONFIG_ANDROID_BINDERFS=y
CONFIG_ANDROID_BINDER_IPC_SELFTEST=y
CONFIG_FUSE_FS=y
CONFIG_FTRACE_SYSCALLS=y

Joshua-Riek, 2024-08-04:
I see these modules are not enabled, but from my understanding you must use a DKMS package for ashmem since it was taken out of the Linux kernel.
```diff
diff --git a/debian.rockchip/config/config.common.ubuntu b/debian.rockchip/config/config.common.ubuntu
index 96ed1c1e2294..fc376b580491 100644
--- a/debian.rockchip/config/config.common.ubuntu
+++ b/debian.rockchip/config/config.common.ubuntu
@@ -147,7 +147,7 @@ CONFIG_RCU_NEED_SEGCBLIST=y
 
 CONFIG_IKCONFIG=y
 CONFIG_IKCONFIG_PROC=y
-# CONFIG_IKHEADERS is not set
+CONFIG_IKHEADERS=y
 CONFIG_LOG_BUF_SHIFT=18
 CONFIG_LOG_CPU_MAX_BUF_SHIFT=12
 CONFIG_PRINTK_SAFE_LOG_BUF_SHIFT=13
@@ -3177,7 +3177,7 @@ CONFIG_WLAN_UWE5622=m
 CONFIG_SPRDWL_NG=m
 CONFIG_UNISOC_WIFI_PS=y
 CONFIG_TTY_OVERY_SDIO=m
-# CONFIG_MAC80211_HWSIM is not set
+CONFIG_MAC80211_HWSIM=y
 CONFIG_USB_NET_RNDIS_WLAN=y
 # CONFIG_VIRT_WIFI is not set
 # CONFIG_WAN is not set
@@ -8350,7 +8350,7 @@ CONFIG_ARM_PMU=y
 CONFIG_ANDROID_BINDER_IPC=y
 CONFIG_ANDROID_BINDERFS=y
 CONFIG_ANDROID_BINDER_DEVICES="binder,hwbinder,vndbinder,anbox-binder,anbox-hwbinder,anbox-vndbinder"
-# CONFIG_ANDROID_BINDER_IPC_SELFTEST is not set
+CONFIG_ANDROID_BINDER_IPC_SELFTEST=y
 # end of Android
 
 # CONFIG_LIBNVDIMM is not set
@@ -9482,7 +9482,7 @@ CONFIG_DYNAMIC_FTRACE_WITH_REGS=y
 # CONFIG_HWLAT_TRACER is not set
 # CONFIG_OSNOISE_TRACER is not set
 # CONFIG_TIMERLAT_TRACER is not set
-# CONFIG_FTRACE_SYSCALLS is not set
+CONFIG_FTRACE_SYSCALLS=y
 # CONFIG_TRACER_SNAPSHOT is not set
 CONFIG_BRANCH_PROFILE_NONE=y
 # CONFIG_PROFILE_ANNOTATED_BRANCHES is not set
```

nrpetonr, 2024-08-04:
try this one, working nicely, even with hw acceleration. u must use 5.10 kernel.

https://github.com/CNflysky/redroid-rk3588

KylinDemons, 2024-08-05:
You can not run Redroid with GPU acceleration in Ubuntu 24.04 kernel 6.1, now it's only suppport Kernel 5.10 on rk3588.

closed 2025-02-10 by github-actions[bot]

Joshua-Riek, 2026-02-02:
Not possible.