Joshua-Riek/ubuntu-rockchip #968: Bug Report: orange pi 5 official wifi module ap6275p not getting loaded

moo-the-cow, 2024-08-04:
### What happened?

U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/orangepi-5-ap6275p.dtbo"
does not detect the module, it worked with the original orange pi 5 ubuntu version by the distributor

### Kernel version

6.1.0-1021-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
dmesg | grep wifi
[   19.570403] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6275p
[   19.570408] [WLAN_RFKILL]: wlan_platdata_parse_dt: enable wifi power control.
[   19.570413] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[   19.570456] [WLAN_RFKILL]: wlan_platdata_parse_dt: The ref_wifi_clk not found !
[   19.570466] [WLAN_RFKILL]: rfkill_set_wifi_bt_power: 1
```

moo-the-cow, 2024-08-04:
I found this, but the question is: was it ever working?
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/315

moo-the-cow, 2024-08-18:
updated to 6.1.0-1023-rockchip

cat /etc/default/u-boot
```
## /etc/default/u-boot - configuration file for u-boot-update(8)

#U_BOOT_UPDATE="true"

#U_BOOT_ALTERNATIVES="default recovery"
#U_BOOT_DEFAULT="l0"
#U_BOOT_PROMPT="1"
#U_BOOT_ENTRIES="all"
#U_BOOT_MENU_LABEL="Debian GNU/Linux"
#U_BOOT_PARAMETERS="ro earlycon"
#U_BOOT_ROOT=""
#U_BOOT_TIMEOUT="50"
U_BOOT_FDT="device-tree/rockchip/rk3588s-orangepi-5.dtb"
#U_BOOT_FDT_DIR="/lib/firmware/"
#U_BOOT_FDT_OVERLAYS=""
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/orangepi-5-ap6275p.dtbo device-tree/rockchip/overlay/orangepi-5-sata.dtbo device-tree/rockchip/overlay/orangepi-5-disable-led.dtbo"
#U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/"
#U_BOOT_SYNC_DTBS="false"
```

maybe some more logs from syslog:
```
2024-08-18T19:16:22.730446+00:00 localhost systemd[1]: Stopping ap6275p-bluetooth.service - Bluetooth AP6275P...
2024-08-18T19:16:47.809867+00:00 localhost systemd[1]: Starting ap6275p-bluetooth.service - Bluetooth AP6275P...
2024-08-18T19:16:47.811883+00:00 localhost kernel: [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6275p
2024-08-18T19:16:52.241866+00:00 localhost ap6275p-bluetooth.sh[1049]: /usr/lib/scripts/ap6275p-bluetooth.sh: line 3: warning: command substitution: ignored null byte in input
2024-08-18T19:16:52.245684+00:00 localhost ap6275p-bluetooth.sh[1049]: /usr/lib/scripts/ap6275p-bluetooth.sh: line 4: warning: command substitution: ignored null byte in input
2024-08-18T19:16:52.359750+00:00 localhost systemd[1]: Started ap6275p-bluetooth.service - Bluetooth AP6275P.
2024-08-18T19:16:52.385239+00:00 localhost ap6275p-bluetooth.sh[1115]: ###AMPAK FW Auto detection patch version = [v1.1 20161117]###
2024-08-18T19:16:52.385581+00:00 localhost ap6275p-bluetooth.sh[1115]: FW folder path = /lib/firmware/ap6275p
2024-08-18T19:16:52.390599+00:00 localhost ap6275p-bluetooth.sh[1115]: proc_resetchip id = BCM4362A2
2024-08-18T19:16:52.390611+00:00 localhost ap6275p-bluetooth.sh[1115]: FW path = /lib/firmware/ap6275p/BCM4362A2.hcd
2024-08-18T19:16:54.339875+00:00 localhost ap6275p-bluetooth.sh[1115]: send hci_download_minidriverproc_resetDone setting line discpline
2024-08-18T19:19:34.489533+00:00 localhost systemd[1]: Stopping ap6275p-bluetooth.service - Bluetooth AP6275P...
2024-08-18T19:19:59.332215+00:00 localhost systemd[1]: Starting ap6275p-bluetooth.service - Bluetooth AP6275P...
2024-08-18T19:19:59.334211+00:00 localhost kernel: [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6275p
2024-08-18T19:20:03.641739+00:00 localhost ap6275p-bluetooth.sh[1097]: /usr/lib/scripts/ap6275p-bluetooth.sh: line 3: warning: command substitution: ignored null byte in input
2024-08-18T19:20:03.644999+00:00 localhost ap6275p-bluetooth.sh[1097]: /usr/lib/scripts/ap6275p-bluetooth.sh: line 4: warning: command substitution: ignored null byte in input
2024-08-18T19:20:03.756791+00:00 localhost systemd[1]: Started ap6275p-bluetooth.service - Bluetooth AP6275P.
2024-08-18T19:20:03.785813+00:00 localhost ap6275p-bluetooth.sh[1153]: ###AMPAK FW Auto detection patch version = [v1.1 20161117]###
2024-08-18T19:20:03.785848+00:00 localhost ap6275p-bluetooth.sh[1153]: FW folder path = /lib/firmware/ap6275p
2024-08-18T19:20:03.786576+00:00 localhost ap6275p-bluetooth.sh[1153]: proc_resetchip id = BCM4362A2
2024-08-18T19:20:03.786602+00:00 localhost ap6275p-bluetooth.sh[1153]: FW path = /lib/firmware/ap6275p/BCM4362A2.hcd
2024-08-18T19:20:05.701871+00:00 localhost ap6275p-bluetooth.sh[1153]: send hci_download_minidriverproc_resetDone setting line discpline
```

AdimasBarelang, 2024-08-23:
Hello, I am facing the same problem. Have you solved the problem? After upgrade it into newest kernel my wifi is gone.

tuftsanti, 2024-09-15:
Guess I can't use this image until the wifi/bluetooth card is made available again

Joshua-Riek, 2024-09-15:
Read the wiki and enable the device tree overlay. Or provide relevant logs and reproducible steps of the problem.

moo-the-cow, 2024-09-16:
I'm confused. I provided very detailed logs (unlike many bug reports) and also I think  "device-tree/rockchip/overlay/orangepi-5-ap6275p.dtbo" is clear enough that I am using the overlay. My guess is just that its not working anymore because of the newer kernel

Joshua-Riek, 2024-09-16:
Oh, you cant use the orangepi-5-sata.dtbo with orangepi-5-ap6275p.dtbo.

moo-the-cow, 2024-09-17:
and we have a winner 
ok I wouldn't have guessed that. is there some kind of documentation pointing out which overlays are incompatible to each other?
Also I think I need both, but I will think of something smart, at least it's good to know that the sata dtbo was responsible.
[image]

tuftsanti, 2024-09-18:
I'm confused what you changed here. I had the stock img installed and couldn't start the ap6275p chip.

avatar21, 2024-10-12:
> Oh, you cant use the orangepi-5-sata.dtbo with orangepi-5-ap6275p.dtbo.

yeap this works for me too.
```sh
cat /etc/default/u-boot
## /etc/default/u-boot - configuration file for u-boot-update(8)

#U_BOOT_UPDATE="true"

#U_BOOT_ALTERNATIVES="default recovery"
#U_BOOT_DEFAULT="l0"
#U_BOOT_PROMPT="1"
#U_BOOT_ENTRIES="all"
#U_BOOT_MENU_LABEL="Debian GNU/Linux"
#U_BOOT_PARAMETERS="ro earlycon"
#U_BOOT_ROOT=""
#U_BOOT_TIMEOUT="50"
#U_BOOT_FDT=""
#U_BOOT_FDT_DIR="/lib/firmware/"
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/orangepi-5-ap6275p.dtbo"
#U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/"
#U_BOOT_SYNC_DTBS="false"

```