Joshua-Riek/ubuntu-rockchip #969: Bug RepoUnable to boot from NVME on Orange Pi 5, v2.3.0 24.04rt:

Bluesmanuk, 2024-08-05:
### What happened?

After writing and booting from an SD card, I thn followed the wicki, as before to issue the 2 command required to transfer and boot from NVME.

sudo u-boot-install-mtd
sudo ubuntu-rockchip-install /dev/nvme0n1

The processes run successfully but after removing the SD card and booting, there is zero activity.

I did note from the general changes in v2.3.0 that "Orange Pi 5 boards now use u-boot-radxa-rk3588" was in the description but I am not sure what the exact ramifications of this are with regards to transferring a system to NVME.

### Kernel version

6.1.0-1021.21

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Joshua-Riek, 2024-08-05:
SPL is borked.
```
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x1
CH0 RX Vref:28.9%, TX Vref:19.8%,21.8%
CH1 RX Vref:28.5%, TX Vref:21.8%,21.8%
CH2 RX Vref:27.9%, TX Vref:20.8%,19.8%
CH3 RX Vref:29.7%, TX Vref:21.8%,22.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09 (Jul 30 2024 - 00:38:31)
unrecognized JEDEC id bytes: 00, 00, 00
unknown raw ID 0 0 0
Trying to boot from MMC2
MMC: no card present
mmc_init: -123, time 1
spl: mmc init failed with error: -123
Trying to boot from MMC1
Card did not respond to voltage select!
mmc_init: -95, time 13
spl: mmc init failed with error: -95
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
# Reset the board to bootrom #
```

Joshua-Riek, 2024-08-05:
I just killed my Orange Pi 5 as I was trying to fix this lol. 

I will need a few days as a new device is in the mail.

Bluesmanuk, 2024-08-05:
Ouch. I'm sorry to hear that. I hope that they are sending you one for free for all the work you are doing in spite of the little that they do.

toonvanstrijp, 2024-08-09:
Is there any way I can help solve this? I'm facing the same problem.

Joshua-Riek, 2024-08-09:
I pushed a fix last night, will update the package later tonight.

Bluesmanuk, 2024-08-10:
Ran the latest updates and then ran the 2 commands again.

All was successful and I can boot from NVME.

After the first command there was one response after completion that stated "diff blocks: 353" but there are no issues that I can detect if this is some form of error.

Thank you.

closed 2024-08-10 by Bluesmanuk