Joshua-Riek/ubuntu-rockchip #970: Bug Report: Ubuntu 24 install on nvme fails on opi5max

vignesh2004, 2024-08-05:
### What happened?

Install to and boot from eMMC -> OK
sudo ubuntu-rockchip-install /dev/nvme0n1 -> fails with error:
"
counting files...
Transferring 178482 files (6751 MB) from desktop-rootfs to /dev/nvme0n1p1. Please wait!
Error: in /usr/bin/ubuntu-rockchip-install on line 107
"
Line 107 : 
"rsync -xavrltD --delete / ${mount_point}/desktop-rootfs/ >/dev/null 2>&1"

can't see the nvme anymore after this error. modprobe fails. 

dmesg says:

"
[  279.665875] Aborting journal on device nvme0n1p1-8.
[  279.665893] Buffer I/O error on dev nvme0n1p1, logical block 121667584, lost sync page write
[  279.665892] EXT4-fs error (device nvme0n1p1): ext4_journal_check_start:83: comm kworker/u16:3: Detected aborted journal
[  279.665900] JBD2: I/O error when updating journal superblock for nvme0n1p1-8.
[  279.665934] Buffer I/O error on dev nvme0n1p1, logical block 0, lost sync page write
[  279.665948] EXT4-fs (nvme0n1p1): I/O error while writing superblock
[  279.665960] EXT4-fs (nvme0n1p1): Remounting filesystem read-only
[  279.666528] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990793, error -30)
[  279.669824] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990793, error -30)
[  279.673156] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990793, error -30)
[  279.676565] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990792, error -30)
[  279.680220] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990792, error -30)
[  279.683767] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990792, error -30)
[  279.684120] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990793, error -30)
[  285.958507] EXT4-fs error (device nvme0n1p1): __ext4_get_inode_loc_noinmem:4617: inode #21233665: block 84934688: comm rsync: unable to read itable block
[  285.958690] EXT4-fs error (device nvme0n1p1): __ext4_get_inode_loc_noinmem:4617: inode #21233665: block 84934688: comm rsync: unable to read itable block
"

### Kernel version

6.1.0-1021-rockchip

### SBC model

Orange pi 5 max

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
dmesg says:

"
[  279.665875] Aborting journal on device nvme0n1p1-8.
[  279.665893] Buffer I/O error on dev nvme0n1p1, logical block 121667584, lost sync page write
[  279.665892] EXT4-fs error (device nvme0n1p1): ext4_journal_check_start:83: comm kworker/u16:3: Detected aborted journal
[  279.665900] JBD2: I/O error when updating journal superblock for nvme0n1p1-8.
[  279.665934] Buffer I/O error on dev nvme0n1p1, logical block 0, lost sync page write
[  279.665948] EXT4-fs (nvme0n1p1): I/O error while writing superblock
[  279.665960] EXT4-fs (nvme0n1p1): Remounting filesystem read-only
[  279.666528] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990793, error -30)
[  279.669824] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990793, error -30)
[  279.673156] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990793, error -30)
[  279.676565] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990792, error -30)
[  279.680220] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990792, error -30)
[  279.683767] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990792, error -30)
[  279.684120] EXT4-fs (nvme0n1p1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 15990793, error -30)
[  285.958507] EXT4-fs error (device nvme0n1p1): __ext4_get_inode_loc_noinmem:4617: inode #21233665: block 84934688: comm rsync: unable to read itable block
[  285.958690] EXT4-fs error (device nvme0n1p1): __ext4_get_inode_loc_noinmem:4617: inode #21233665: block 84934688: comm rsync: unable to read itable block
"
```

Joshua-Riek, 2024-08-05:
Looks like an IO error, I would try again after a reboot.

vignesh2004, 2024-08-06:
Used dd instead and got it working! Since I didn't reboot, wipe and test again, this can be closed for now, thanks!

closed 2024-08-06 by vignesh2004