Joshua-Riek/ubuntu-rockchip #971: Info to Dev: Mali, Firefox & ff2mpv addon

defencedog, 2024-08-05:
I have recently seen Joshua video [The best browser on ARM Linux so far](https://m.youtube.com/watch?v=cUDwmaCrKD8&pp=ygUKTWljcm9sdW51eA==)

Chromium addon addition is very imperfect for instance I cannot install _uBlock Origin_ addon on my OrangePi 3b

Recently I was browsing Mesa VPU / GPU acceleration on `X11` instead of `wayland` & I found interesting links on Odroid forum

1- [MPV + Youtube HW Working](https://forum.odroid.com/viewtopic.php?p=331260#p331260)
2- Important stuff for mpv compilation with mali support & how to use it for media consumption. Re: [Archlinux: X GPU Acceleration Guide
](https://forum.odroid.com/viewtopic.php?p=303710#p303710)

I will request @Joshua-Riek to chime in on the data I collected & what is his opinion?

Gi-0n, 2024-08-05:
Hi again just to mention that in 24.04 with chromium 114 I was able to install ublock

defencedog, 2024-08-05:
@Gi-0n which SBC?

Gi-0n, 2024-08-05:
orange pi 3b v2.1

defencedog, 2024-08-05:
@Gi-0n are you using customised `.img` or official Joshua release?

Gi-0n, 2024-08-05:
I'm using Ubuntu 24.04 for this project

defencedog, 2024-08-05:
@Gi-0n your Bluetooth working? ... its very important for me
Mine SBC same as yours Opi3b v2.1

Gi-0n, 2024-08-05:
I answered this in another thread but I will repeat it, Bluetooth does not work, the audio via 3.5mm jack, neither does the VPU or NPU (with Chrome).

defencedog, 2024-08-05:
There is a thread where I referred a Chinese post. Inrecently downloaded official `jammy-server-1.0.6`  & then installed `ubuntu-desktop`. I got gnome but with `X11` & per the Chinese `wayland` will not work for official images.

However an interesting find. After I installed `kodi` I can log into its `wayland` session (as well as `X11` session) ..see the difference the latter supports `HDR10`
[image]

lopirata, 2024-08-05:
> orange pi 3b v2.1

Ethernet is working in 24.04?

Joshua-Riek, 2024-08-05:
This feels so off topic and I really do not care about RK3566.

closed 2024-08-05 by Joshua-Riek