Joshua-Riek/ubuntu-rockchip #972: Bug Report:  WARNING: erroneous pipeline: no element "mpph265enc"

glowfishDE, 2024-08-05:
### What happened?

i cannot get the mpph265enc/264enc to be available. I tried to install the rockchip-multimedia-config and librockchip-mpp1 and librockchip-vpu0 and gstreamer1.0-rockchip1 packages but gst-inspect-1.0 mpph265enc always tell me "No such element or plugin 'mpph265enc'" 

Do you have any idea what i could have done wrong?

Thank you so much.

### Kernel version

6.10.1-edge-rockchip-rk3588

### SBC model

Orange Pi5+

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
gst-inspect-1.0 mpph265enc
No such element or plugin 'mpph265enc'
```

Joshua-Riek, 2024-08-05:
I think you are reporting in the wrong place. The kernel "6.10.1-edge-rockchip-rk3588" is from Armbian and experimental.

glowfishDE, 2024-08-05:
oh my dear - i worked on the wrong SD card :( sorry mate! should it work with your 6.1.0 kernel and 24.04 LTS ?

glowfishDE, 2024-08-05:
I am on 6.1.0-1021-rockchip now as expected. I could install the mpph265enc with apt-get install gstreamer1.0-rockchip1 but now i get the following errors:

`2024-08-06T01:11:01.037345+02:00 localhost mpp[3863]: mpp_info: mpp version: unknown mpp version for missing VCS info
2024-08-06T01:11:01.037620+02:00 localhost mpp[3863]: mpp_platform: client 12 driver is not ready!
2024-08-06T01:11:01.037787+02:00 localhost mpp[3863]: mpp_info: mpp version: unknown mpp version for missing VCS info
2024-08-06T01:11:01.038068+02:00 localhost mpp[3863]: message repeated 2 times: [ mpp_info: mpp version: unknown mpp version for missing VCS info]
2024-08-06T01:11:01.042052+02:00 localhost belaUI[1033]: Max bitrate: 8000000
2024-08-06T01:11:01.042156+02:00 localhost belaUI[1033]: (belacoder:3863): GLib-GObject-CRITICAL **: 01:11:01.041: g_object_set_is_valid_property: object class 'GstMppH265Enc' has no property named 'bitrate'
2024-08-06T01:11:01.042278+02:00 localhost belaUI[1033]: A-V delay: 0 ms
2024-08-06T01:11:01.042385+02:00 localhost belaUI[1033]: Failed to get a delay element from the pipeline, not applying a delay
2024-08-06T01:11:01.042486+02:00 localhost belaUI[1033]: Failed to get a ptsfixup element from the pipeline, not removing PTS jitter
2024-08-06T01:11:01.096791+02:00 localhost belaUI[1033]: 192.168.10.100 (0xaaab24a7d970): connection group registered
2024-08-06T01:11:04.043317+02:00 localhost belaUI[1033]: Failed to establish an SRT connection: connection timed out. Retrying...`

Would be really cool if you could point me in the right direction. I think i am doing something wrong.

My gstreamer pipeline looks like this:

v4l2src device=/dev/hdmirx ! identity name=ptsfixup signal-handoffs=TRUE ! identity drop-buffer-flags=GST_BUFFER_FLAG_DROPPABLE !
identity name=v_delay signal-handoffs=TRUE !
textoverlay text='' valignment=top halignment=right font-desc="Monospace, 5" name=overlay ! queue !
mpph265enc zero-copy-pkt=0 qp-max=51 gop=120 name=venc_bps !
h265parse config-interval=-1 ! queue max-size-time=10000000000 max-size-buffers=1000 max-size-bytes=41943040 ! mux.
alsasrc device=hw:CARD=rockchiphdmiin ! identity name=a_delay signal-handoffs=TRUE ! volume volume=1.0 !
audioconvert ! voaacenc bitrate=128000 ! aacparse ! queue max-size-time=10000000000 max-size-buffers=1000 ! mux.
mpegtsmux name=mux !
appsink name=appsink

Joshua-Riek, 2024-08-05:
I'm not very familiar with gstreamer, but take a look at this comment https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/199#discussioncomment-6299483.

closed 2025-02-10 by github-actions[bot]