Joshua-Riek/ubuntu-rockchip #973: Feature Request: Board config scripts should update /etc/deault/u-boot

shaunco, 2024-08-06:
### Feature Description

Currently `/etc/default/u-boot` is completely blank:
```
## /etc/default/u-boot - configuration file for u-boot-update(8)

#U_BOOT_UPDATE="true"

#U_BOOT_ALTERNATIVES="default recovery"
#U_BOOT_DEFAULT="l0"
#U_BOOT_PROMPT="1"
#U_BOOT_ENTRIES="all"
#U_BOOT_MENU_LABEL="Debian GNU/Linux"
#U_BOOT_PARAMETERS="ro earlycon"
#U_BOOT_ROOT=""
#U_BOOT_TIMEOUT="50"
#U_BOOT_FDT=""
#U_BOOT_FDT_DIR="/lib/firmware/"
#U_BOOT_FDT_OVERLAYS=""
#U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/"
#U_BOOT_SYNC_DTBS="false"
```

But when a script like [`roc-rk3588s-pc.sh`](https://github.com/Joshua-Riek/ubuntu-rockchip/blob/main/config/boards/roc-rk3588s-pc.sh) runs, this file should be updated with all the build-time settings for the ROC-RK3588S-PC board so that it is easier to make a minor change and rerun `u-boot-update`.

shaunco, 2024-08-06:
Also, am I losing my mind or does `u-boot-update` really not support `overlay_prefix`???

Joshua-Riek, 2024-08-06:
Please read the manpage of u-boot-menu.

> In order to avoid situations where the system might be rendered unbootable  by  a  package
       removal,  other  packages and downstream distributions MUST NOT edit or modify files under
       [/etc](file:///etc) as those are meant for use by end-users  only.  Instead,  it  is  mandated  for  such
       packages to install their config fragments ONLY under /usr/share/u-boot-menu/conf.d/.

Ref: https://manpages.ubuntu.com/manpages/noble/man8/u-boot-update.8.html

Joshua-Riek, 2024-08-06:
> Also, am I losing my mind or does `u-boot-update` really not support `overlay_prefix`???

Correct, why would it support `overlay_prefix`?

closed 2024-08-06 by Joshua-Riek