Joshua-Riek/ubuntu-rockchip #974: USB_HID

kalyanikasam, 2024-08-06:
Hello Joshua,
I have noticed that USB_HID is built as a module in your configuration Ubuntu22.04 CONFIG_USB_HID=y, how can i change the parameter to 'm', as i am connecting a USB touch screen to Rock5B and was unable to get the touch functionality working until i manually map input to corresponding output which will happen only after user login.
I have tried connecting the same display with radxa image, and the touch is working fine. I have gone compared the config files in both OS's and i have noticed that  CONFIG_USB_HID=m in radxa image. Is this the issue. Please guide

Joshua-Riek, 2024-08-06:
Could you provide an example of how USB_HID is used as a module? 

From my understanding all the radxa touch screens and displays now work perfectly.

kalyanikasam, 2024-08-07:
This display is not from radxa it is a 7inch touch screen from waveshare which uses HDMI for display out and USB for touch interface https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C). 
I have compared the configuration file from Ubuntun22.04 in /boot/config-5.10.0-1007-rockchip which says 
**CONFIG_HID_MULTITOUCH=m**
and in config file in radxa image /boot/config-6.1.43-14-rk2312   **CONFIG_HID_MULTITOUCH=y** , Is this why my touch is not working until i manually map HDMI out to USB input using **map-to-output**

closed 2024-12-16 by github-actions[bot]