Joshua-Riek/ubuntu-rockchip #976: Rock5b plus Can't see onboard EMMC device(s)

pascal71, 2024-08-06:
### What happened?

Installed Riek's Ubuntu 24.04 (server edition) on NVME (could'nt get installed using rkdeveloptool). It starts up. But when I try to find the EMMC devices, there are none with lsblk.

Only:

pascal@rock5bp-n01:~$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0   94M  1 loop /snap/lxd/29579
loop1         7:1    0 69.2M  1 loop /snap/core22/1383
loop2         7:2    0 33.7M  1 loop /snap/snapd/21761
loop3         7:3    0 69.2M  1 loop /snap/core22/1438
loop4         7:4    0 94.5M  1 loop /snap/lxd/29733
mtdblock0    31:0    0   16M  0 disk 
nvme1n1     259:0    0  3.7T  0 disk 
├─nvme1n1p1 259:1    0    4M  0 part 
└─nvme1n1p2 259:2    0  3.7T  0 part /
nvme0n1     259:3    0  1.9T  0 disk 

I'd expected to see the EMMC device.

This is running on the RADXA Rock5b-plus., 24GiB RAM edition

### Kernel version

6.1.0-1021

### SBC model

Radxa ROCK 5B Plus 24GiB 

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

pascal71, 2024-08-06:
Some more logs:

[   14.911231] dwmmc_rockchip fe2c0000.mmc: No normal pinctrl state
[   14.911239] dwmmc_rockchip fe2c0000.mmc: No idle pinctrl state

[   14.911288] dwmmc_rockchip fe2c0000.mmc: IDMAC supports 32-bit address mode.
[   14.911300] dwmmc_rockchip fe2c0000.mmc: Using internal DMA controller.
[   14.911307] dwmmc_rockchip fe2c0000.mmc: Version ID is 270a

[   14.911359] dwmmc_rockchip fe2c0000.mmc: Looking up vmmc-supply from device tree
[   14.911551] sdhci-dwcmshc fe2e0000.mmc: Looking up vmmc-supply from device tree

[   14.911828] mmc0: CQHCI version 5.10
[   14.924374] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[   14.941197] mmc0: SDHCI controller on fe2e0000.mmc [fe2e0000.mmc] using ADMA

Will this help?

pascal71, 2024-08-06:
pascal@rock5bp-n01:~$ sudo dmesg | grep mmc
[   14.911231] dwmmc_rockchip fe2c0000.mmc: No normal pinctrl state
[   14.911239] dwmmc_rockchip fe2c0000.mmc: No idle pinctrl state
[   14.911288] dwmmc_rockchip fe2c0000.mmc: IDMAC supports 32-bit address mode.
[   14.911300] dwmmc_rockchip fe2c0000.mmc: Using internal DMA controller.
[   14.911307] dwmmc_rockchip fe2c0000.mmc: Version ID is 270a
[   14.911329] dwmmc_rockchip fe2c0000.mmc: DW MMC controller at irq 97,32 bit host data width,256 deep fifo
[   14.911359] dwmmc_rockchip fe2c0000.mmc: Looking up vmmc-supply from device tree
[   14.911551] sdhci-dwcmshc fe2e0000.mmc: Looking up vmmc-supply from device tree
[   14.911552] dwmmc_rockchip fe2c0000.mmc: Looking up vqmmc-supply from device tree
[   14.911567] sdhci-dwcmshc fe2e0000.mmc: Looking up vmmc-supply property in node /mmc@fe2e0000 failed
[   14.911585] sdhci-dwcmshc fe2e0000.mmc: Looking up vqmmc-supply from device tree
[   14.911587] sdhci-dwcmshc fe2e0000.mmc: Looking up vqmmc-supply property in node /mmc@fe2e0000 failed
[   14.911828] mmc0: CQHCI version 5.10
[   14.924374] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[   14.941197] mmc0: SDHCI controller on fe2e0000.mmc [fe2e0000.mmc] using ADMA

pascal71, 2024-08-08:
@hanvinke

You are right. The EMMC component is unsoldered on the board; which is probably the hardware equivalent of 'vapourware'. 
I actually bought 3 of these for the onboard EMMC but alas.. :)

No worries; this issue can be closed. What is not there, can not be found. 

I will open another issue for the POE hat that doesn't want it's fan to spin, even when I instruct it do so manually.

pascal71, 2024-08-08:
Issue can be closed as there's no EMMC soldered on the current Rock5b-plus boards.

abratchik, 2024-08-08:
I have the similar problem. My board is Orange Pi5 Plus 16GB + 256Gb eMMC module installed. Normally the eMMC module is visible under /dev/mmcblk1 name but I see the SD card under this name only. eMMC is not visible at all so dd on it is not possible.

hanvinke, 2024-08-08:
@pascal71 
No worries too - it is still a wonderfull board. Flawless 4K mpp youtube and 10 bit HDR X265 video play, it all works perfectly! So no regrets buying it.

pascal71, 2024-08-09:
No worries. Now onto getting POE 25W 's fan running
Next issue :)

pascal71, 2025-01-20:
This one can be safely closed. RADXA appears to advertise with with- but ships them without EMMC

closed 2025-01-20 by pascal71