Joshua-Riek/ubuntu-rockchip #978: Bug Report: Rock 5B CAN enable

vzhikh, 2024-08-07:
### What happened?

Hi,
Failed to enable CAN1 with overlay:
`U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-can1-m1.dtbo"`

U-boot successfully updated:
```
ubuntu@ubunturock5b:~$ cat /boot/extlinux/extlinux.conf | grep "can1"
	fdtoverlays  /lib/firmware/6.1.0-1021-rockchip/device-tree/rockchip/overlay/rk3588-can1-m1.dtbo 
```
File successfully loaded on boot:
```
Retrieving file: /lib/firmware/6.1.0-1021-rockchip/device-tree/rockchip/overlay/rk3588-can1-m1.dtbo
343 bytes read in 241 ms (1000 Bytes/s)

```

But there is no can device in system:
```
ubuntu@ubunturock5b:~$ sudo ifconfig -a
enP4p65s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.10.0.176  netmask 255.255.252.0  broadcast 10.10.3.255
        inet6 fe80::2e0:4cff:fe03:2b0b  prefixlen 64  scopeid 0x20<link>
        ether 00:e0:4c:03:2b:0b  txqueuelen 1000  (Ethernet)
        RX packets 30897  bytes 2189841 (2.1 MB)
        RX errors 0  dropped 192  overruns 0  frame 0
        TX packets 495  bytes 102416 (102.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 92  bytes 7159 (7.1 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 92  bytes 7159 (7.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

### Kernel version

6.1.0-1021-rockchip

### SBC model

Rock 5B

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

vzhikh, 2024-08-08:
On 22.04 it works

Joshua-Riek, 2024-08-25:
The Rockchip CAN driver was not fully ported to Linux 6.1. I added a few changes and it should work as expected. 

https://github.com/Joshua-Riek/linux-rockchip/commit/acf81a30451ddf8df672da41d60ed5917d98a0c8
https://github.com/Joshua-Riek/linux-rockchip/commit/685cba94b3618660b2a56de0b2cb9d0f6250cc13

closed 2025-02-10 by github-actions[bot]