Joshua-Riek/ubuntu-rockchip #979: Question: MPV Option Transcode 4K UHD HEVC to H264? Orange Pi 5

conrad10781, 2024-08-07:
Hello. Thank you for work being done this repository. 

I've run into an issue with MPV when the source is 4K UHD HEVC and the video stutters on playback.

I saw in https://github.com/Joshua-Riek/ubuntu-rockchip/issues/345#issuecomment-1704336977 

_"The Wayland display manager does not support HDR video playback. The alternative is to use the X11 display manager, but it has very bad performance and a lot of graphical glitches."_

While that comment is from September 2023, presuming it's still valid, I've tried in Wayland and X11 with various MPV parameters, but the result is the same. I've also tried both ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5 and ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5 with the same result.

My videos were all made with MakeMKV using the 4K UHD movies in my collection. I didn't realize it at the time, but MakeMKV appears to use HEVC by default.

In any event, if I re-encode any of the videos to H264 with something like below, the video then plays fine.

ffmpeg -i input.mkv -map 0 -c:v libx264 -crf 18 -vf format=yuv420p -c:a copy output.mkv

This is all a long-winded way of asking if there is maybe a way to transcode the video to H264 as I'm watching it, so that I don't have to re-encode my entire collection, and maybe more importantly need twice the storage and have duplicates of every video in the collection? Especially given it's unlikely every single title would be watched any time in the near future.

Joshua-Riek, 2024-08-08:
Can you provide a download link to a video i can test with? Also what was the exact mpv command you used?

conrad10781, 2024-08-09:
[3-seconds.zip](https://github.com/user-attachments/files/16561420/3-seconds.zip)

^-- Here is 3 seconds of the video created using:

ffmpeg -ss 00:10:00 -i input -t 00:00:03 -map 0 -c copy 3-seconds.mkv

Anything larger than that I can't upload here, and I'm a little hesitant to post on a public link and have people think I'm pirating a movie, but am open to alternatives if you have them.

For mpv commands, I tried various combinations of:

--wayland-disable-vsync=yes
--hdr-compute-peak=no
--hwdec=auto
--hwdec-codecs=all

Possibly some others, those are what I see in the history, but I had re-installed the OS previously and may have tried some others that I don't have a reference to anymore.

For further clarification, on top of the video stuttering, the console will report a growing number of dropped frames. To the tune of ~120 frames for the 3 second video I provided.

Here is output from a fresh install just using mpv with no other parameters I just did. I don't always get the HEVC errors that are in there. This was just from a few seconds of the video playing. 

[image]

Also, for what it's worth, I get the issues whether the file is local on the SD card or on the NFS mount. I can play the video fine without issues on my desktop on the same network.