Joshua-Riek/ubuntu-rockchip #980: Bug Report: Radxa 25W-POE HAT's fan is not working on rock-5b-plus

pascal71, 2024-08-09:
### What happened?

Installed ubunt 24.04 version on rock-5b-plus. I have mounted the Radxa 25W POE HAT on it. The device gets power, boots up nicely, but the fan is never spinning up. Not even when forcing it's PWM fan to spin.

It appears there is a new 25W POE overlay for that supports both ROCK5B and ROCK5B PLUS in the b2 debian version of the Radxa debian image. However, I do not know how to compile it against Ubuntu 24.04.

  

### Kernel version

6.1.0-1021-rockchip

### SBC model

radxa-rock-5b-plus

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

pascal71, 2024-08-09:
root@rock5bp-n01:/tmp/usr/src/radxa-overlays-0.1.13/arch/arm64/boot/dts/rockchip/overlays# cat rock-5b-radxa-25w-poe.dts 
/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/pinctrl/rockchip.h>
#include <dt-bindings/pwm/pwm.h>
#include <dt-bindings/thermal/thermal.h>

/ {
	metadata {
		title = "Enable Radxa 25W PoE+ HAT";
		category = "misc";
		compatible = "radxa,rock-5b", "radxa,rock-5b-plus";
		description = "Enable Radxa 25W PoE+ HAT.";
		exclusive = "GPIO3_A7", "GPIO3_C3";
		package = "rsetup-config-thermal-governor-step-wise";
	};
};

&{/} {
	radxa_pow_w1: radxa-poe-w1 {
		compatible = "w1-gpio";
		gpios = <&gpio3 RK_PC3 GPIO_ACTIVE_HIGH>;
		status = "okay";
	};

	radxa_poe_fan: radxa-poe-fan {
		compatible = "pwm-fan";
		#cooling-cells = <2>;
		cooling-min-state = <0>;
		cooling-max-state = <4>;
		cooling-levels = <0 64 128 192 255>;
		pwms = <&pwm8 0 40000 PWM_POLARITY_INVERTED>;
	};
};

&pwm8 {
	status = "okay";
	pinctrl-0 = <&pwm8m0_pins>;
};

&soc_thermal {
	cooling-maps {
		map4 {
			cooling-device = <&fan0 THERMAL_NO_LIMIT THERMAL_NO_LIMIT>,
								<&radxa_poe_fan THERMAL_NO_LIMIT THERMAL_NO_LIMIT>;
		};
		map5 {
			cooling-device = <&fan0 THERMAL_NO_LIMIT THERMAL_NO_LIMIT>,
								<&radxa_poe_fan THERMAL_NO_LIMIT THERMAL_NO_LIMIT>;
		};
	};
};
root@rock5bp-n01:/tmp/usr/src/radxa-overlays-0.1.13/arch/arm64/boot/dts/rockchip/overlays#

nascs, 2024-08-13:
you can fefer to [this dts](https://github.com/radxa-pkg/radxa-overlays/blob/main/arch/arm64/boot/dts/rockchip/overlays/rock-5b-radxa-25w-poe.dts).

nascs, 2024-08-13:
1. compile dts to dtbo
2. copy dtbo to overlay directory by the cmd  "cp rock-5b-radxa-25w-poe.dtbo /lib/firmware/5.10.0-1011-rockchip/device-tree/rockchip/overlay/rock-5b-radxa-25w-poe.dtbo"
3. add a new line "fdtoverlays  /lib/firmware/5.10.0-1011-rockchip/device-tree/rockchip/overlay/rock-5b-radxa-25w-poe.dtbo" in the file /boot/extlinux/extlinux.conf
4. reboot your system

pascal71, 2024-08-13:
Thanks William;

But.. I probably need the source code of the kernel from rockchip  
ubuntu 24.04. Can you share with me which packages I need. (The dts  
file refers to some include files I cannot find)

Thanks in advance.

I will document the procedure once I have it working.

Kind regards,

    Pascal van Dam

nascs, 2024-08-13:
I recommend that you can use the [radxa-overlays](https://github.com/radxa-pkg/radxa-overlays) repository to compile overlay.

nascs, 2024-08-13:
I'll write a tutorial on how to use the radxa-overlays repository later on.

Joshua-Riek, 2024-08-13:
The device tree for the Rock 5B in Ubuntu Rockchip has a different cooling map configuration, I need to take a look to see how to make the overlay work.

nascs, 2024-08-13:
> I'll write a tutorial on how to use the radxa-overlays repository later on.

Hi @pascal71, for overlay compilation, you can refer to [this guide](https://docs.radxa.com/en/rock5/rock5b/low-level-dev/overlays).

pascal71, 2024-08-14:
Many thanks for your response.

For Ubuntu 24.04 the 6.x kernel is used, so I probably need that kernel source/headers instead of the 5.x one. Correct? Next question; where to get that one?

nascs, 2024-08-14:
For Ubuntu 24.04 the 6.x kernel source, maybe you can refer to [noble.sh](https://github.com/Joshua-Riek/ubuntu-rockchip/blob/main/config/suites/noble.sh).

tyrellm04, 2024-08-19:
I'm also attempting to get the PoE hat fan spinning, however, when I attempt to compile the rock-5b-radxa-25w-poe.dts, I get a parsing error: 
Error: rock-5b-radxa-25w-poe.dts:4.1-9 syntax error
FATAL ERROR: Unable to parse input tree

I'm running DCT 1.7.0. When I view the file in VS Code with the DST formatter it seems to pass muster there.

pascal71, 2025-01-20:
Let me try to pick this one up