Joshua-Riek/ubuntu-rockchip #982: Bug Report: NanoPi R6S shows only a black screen after boot

Blind55, 2024-08-10:
### What happened?

Upon boot, the bootloader (Ubuntu) screen shows up with its spinning wheels.
However, subsequently, the login manager only shows a "black screen" (the mouse cursor, however, is visible, and movable).
Switching to the console with alt-ctrl-F4 shows a normal login prompt.
Switching to the console and back to the graphical environment lets one for a brief fraction of a second see the 'create your account' window, but immediately the screen turns back to black.

Using:
https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.3.0/ubuntu-24.04-preinstalled-desktop-arm64-nanopi-r6s.img.xz

### Kernel version

Rockchip Linux 6.1

### SBC model

NanoPi R6S

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Joshua-Riek, 2024-08-11:
What resolution are you using?

Blind55, 2024-08-11:
I have tried standard HD (1920x1080) and UHD (3840x2160).
I was trying to adjust the gdm3 config in /etc/gdm/custom.conf and switch to xorg instead of wayland, but that didn't seem to change anything. I have the feeling if I blindly set the mouse to the right locations, the GUI would let me create the account - if I could see the windows ... they appear to be functional.
In summary, something ('black') seems to be overlaying the (functional) window system.

Blind55, 2024-08-23:
A few further comments:
1. 22.04 seems to have the same issue
2. I cannot really do much with the system online as there is no account I can log into, I think
    (makes it hard to see the logs, really)
3. I can, however, try to turn off services having mounted the partition in a different system. It would be helpful to know where I could start, if there is a target service I could try.
4. The newest release appears to not open the window for account creation, as far as I can glean from the very brief switches from console to graphical interface (before the graphical interface goes black)

Zenkala, 2024-09-03:
Same issue here (same Ubuntu version, but a NanoPi R6C).
I thought moving from gdm3 to lightdm worked, since I got the desktop to show. But as soon as I move the mouse, it's gone again. Interaction still works (mostly keyboard).
I suspect it is more a GPU driver issue than a window manager issue.

```
Aug 15 23:59:14 localhost.localdomain kernel: rockchip-drm display-subsystem: failed to parse loader memory
Aug 15 23:59:14 localhost.localdomain kernel: ------------[ cut here ]------------
Aug 15 23:59:14 localhost.localdomain kernel: Must have as many primary planes as there are CRTCs, but have 0 primary planes and 1 CRTCs
Aug 15 23:59:14 localhost.localdomain kernel: WARNING: CPU: 4 PID: 1 at drivers/gpu/drm/drm_mode_config.c:692 drm_mode_config_validate+0xdc/0x360
Aug 15 23:59:14 localhost.localdomain kernel: Modules linked in:
Aug 15 23:59:14 localhost.localdomain kernel: CPU: 4 PID: 1 Comm: swapper/0 Tainted: G        W          6.1.0-1023-rockchip #23-Ubuntu
Aug 15 23:59:14 localhost.localdomain kernel: Hardware name: FriendlyElec NanoPi R6C (DT)
Aug 15 23:59:14 localhost.localdomain kernel: pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 15 23:59:14 localhost.localdomain kernel: pc : drm_mode_config_validate+0xdc/0x360
Aug 15 23:59:14 localhost.localdomain kernel: lr : drm_mode_config_validate+0xdc/0x360
Aug 15 23:59:14 localhost.localdomain kernel: sp : ffff80000b45b960
Aug 15 23:59:14 localhost.localdomain kernel: x29: ffff80000b45b960 x28: 0000000000000001 x27: ffff80000b3d6088
Aug 15 23:59:14 localhost.localdomain kernel: x26: ffff80000b1a7808 x25: ffff0001f2979c48 x24: ffff800009d8a95f
Aug 15 23:59:14 localhost.localdomain kernel: x23: ffff800009d8a91a x22: 0000000000000002 x21: 0000000000000002
Aug 15 23:59:14 localhost.localdomain kernel: x20: ffff0001f2979800 x19: ffff0001f2979c38 x18: 0000000000000000
Aug 15 23:59:14 localhost.localdomain kernel: x17: 0000000000000000 x16: 0000000000000000 x15: 000000000000000a
Aug 15 23:59:14 localhost.localdomain kernel: x14: 0000000000000001 x13: ffffffffffffffff x12: ffffffffffffffff
Aug 15 23:59:14 localhost.localdomain kernel: x11: 0000000000000000 x10: ffff80000b3569f3 x9 : ffff8000080ea040
Aug 15 23:59:14 localhost.localdomain kernel: x8 : 0000000073435452 x7 : 2073656e616c7020 x6 : 0000000054524320
Aug 15 23:59:14 localhost.localdomain kernel: x5 : ffff80000b356999 x4 : 0000000000000000 x3 : 0000000000000000
Aug 15 23:59:14 localhost.localdomain kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff0001f0848000
Aug 15 23:59:14 localhost.localdomain kernel: Call trace:
Aug 15 23:59:14 localhost.localdomain kernel:  drm_mode_config_validate+0xdc/0x360
Aug 15 23:59:14 localhost.localdomain kernel:  drm_dev_register+0x38/0x1f0
Aug 15 23:59:14 localhost.localdomain kernel:  rockchip_drm_bind.part.0+0x404/0x548
Aug 15 23:59:14 localhost.localdomain kernel:  rockchip_drm_bind+0x74/0x7c
Aug 15 23:59:14 localhost.localdomain kernel:  try_to_bring_up_aggregate_device+0xf0/0x140
Aug 15 23:59:14 localhost.localdomain kernel:  component_master_add_with_match+0xb0/0xec
Aug 15 23:59:14 localhost.localdomain kernel:  rockchip_drm_platform_probe+0x270/0x2ac
Aug 15 23:59:14 localhost.localdomain kernel:  platform_probe+0x70/0xc0
Aug 15 23:59:14 localhost.localdomain kernel:  really_probe+0x1cc/0x390
Aug 15 23:59:14 localhost.localdomain kernel:  __driver_probe_device+0x13c/0x154
Aug 15 23:59:14 localhost.localdomain kernel:  driver_probe_device+0x48/0xd0
Aug 15 23:59:14 localhost.localdomain kernel:  __driver_attach+0x168/0x178
Aug 15 23:59:14 localhost.localdomain kernel:  bus_for_each_dev+0x7c/0xcc
Aug 15 23:59:14 localhost.localdomain kernel:  driver_attach+0x2c/0x38
Aug 15 23:59:14 localhost.localdomain kernel:  bus_add_driver+0xf8/0x20c
Aug 15 23:59:14 localhost.localdomain kernel:  driver_register+0xbc/0xf8
Aug 15 23:59:14 localhost.localdomain kernel:  __platform_driver_register+0x30/0x3c
Aug 15 23:59:14 localhost.localdomain kernel:  rockchip_drm_init+0xe4/0x120
Aug 15 23:59:14 localhost.localdomain kernel:  do_one_initcall+0x80/0x1c0
Aug 15 23:59:14 localhost.localdomain kernel:  kernel_init_freeable+0x298/0x2a0
Aug 15 23:59:14 localhost.localdomain kernel:  kernel_init+0x30/0x140
Aug 15 23:59:14 localhost.localdomain kernel:  ret_from_fork+0x10/0x20
Aug 15 23:59:14 localhost.localdomain kernel:
```

Blind55, 2024-09-25:
Anything that could be done by me in support of fixing this bug?

closed 2025-02-10 by github-actions[bot]