Joshua-Riek/ubuntu-rockchip #984: Bug Report: Ubuntu server 24.04 won't boot on Orange Pi 3B v2.1

DiegoBM, 2024-08-11:
### What happened?

I'm not sure if this could be related to [this ](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/965) issue, but the ubuntu server 24.04 image won't boot at all in my orange pi 3B v2.1.

I don't know either if it might be related to this, but in my case I did install the manufacturer's ubuntu server image in the NVME and SPI-flash about a month ago, which was working fine, and then I found out about this distro so I wanted to switch to it. I did burn Joshua's 24.04 ubuntu server image into a microsdxc card using Balena etcher and tried to boot from it previously removing the NVME card to ensure that it boots from TF. I can see the activity led blinking but nothing coming out to hdmi. The board won't get an ip address either connected to ethernet, to try to ssh into it, so I presume it's not booting.

### Kernel version

?

### SBC model

Orange Pi 3B v2.1

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

DiegoBM, 2024-08-19:
Noone usgin 3B with this distro?

Gi-0n, 2024-08-23:
Hi, I have the Orange Pi 3b (V2.1) but I have not used it in server mode, in desktop mode it does load the system.

DiegoBM, 2024-08-23:
Thank you for your reply @Gi-0n

Yes, the 24.04 desktop flavor is the only one that seems to work fine. I wonder what would be the difference with the other 3. In any case, since I'm only interested in using it headless, I'm afraid this distro won't work..

Gi-0n, 2024-08-24:
Today I tried to download an image of Ubuntu 24.04 server and it did load the system. I attach a photo of my screen running the system.
[image]

Gi-0n, 2024-08-24:
@DiegoBM I Download the image ubuntu-server-24.04 (v2.3.1) and use the Chrome USB Recovery to flash it to the micro Sd (my only card reader is a chromebook :P) I hope this information helps you

DiegoBM, 2024-08-24:
Thank you @Gi-0n, I didn't try the latest release, since there was no reaction to this issue I assumed nothing would have changed on the 3B side, but I'll give it a spin. I cannot use that card writer though, I hope that's not the problem

DiegoBM, 2024-08-25:
@Gi-0n unfortunately I keep getting the same results with the new release so I set myself to investigate a bit and [this armbian post](https://forum.armbian.com/topic/42698-armbian-on-orange-pi-3b-with-vendor-images-linux-66-dont-seem-to-work/#comment-197222) set me in some sort of direction direction (I didn't read the entire thing). I was experiencing the same results as the original poster, and the same armbian versions worked and failed for me, so it seems like they must have done something on the orange pi side for the newer models (all my cards are at most about one month old).

Whatever is going on in my case seems to be HDMI related and I've found that only @Joshua-Riek 's 24.04 desktop version seems to be able to connect to HDMI consistently. For 24.04 server version I've found that if the monitor is connected when I turn on the orange pi, it won't display anything, but if I turn on the monitor about a second after the green light starts blinking, then it will start working. If I wait too long it won't display anything either, there seems to be a window there...

In any case I've found a very minimalistic distribution called DietPi which seems to work flawlessly, so I'll probably stick to that for my 3B, and I might continue using Joshua's for my 5 max, depending on my experience with DietPi, but that one sure has a tiny footprint!

igorpecovnik, 2024-08-29:
> same armbian versions worked and failed for me

Its community maintained: https://www.armbian.com/orangepi3b/

hw is broken, there are at least two different revisions on the market, so costs to support are higher and Orangepi have absolute no intention to support developers. There are other vendors and other products, so its better choosing something else. Costs of boards are usually lower then 1h engineer hour and all requires docents of hours. Difference is not worth spending time talking about. 

Real support is hard and expensive work. For Dietpi guy, "support" means downloading image from someone, including support. Mainly he is downloading Armbian and then auto-renaming it to Dietpi. In case of Opi3b he probably re-branded original (also an Armbian fork with "search for Armbian replace with Orangepi") OrangeOS with smelly promo & unmaintained Rockchip kernel (but where hw functions works well) to dietpi. He only cares to make as many images as possible to "support" you.

When something serious comes up or something that needs time to fix, he always tries all ways to push responsibility to 3rd party developers. He in 10 years didn't manage to contribute worth mention common (Armbian or up) value he is selling under different name to you. There were several trivial changes, so I can't use word "nothing". Only where and when he could gain points in front of his users with absolute minimum efforts, then he contributed to common. He also solicitate his end-user to help him making a pressure to developers, which are without his greed already overloaded.

Only in case something doesn't work, then you hear about projects where he takes the value from, always in negative context. Smart? Yes. Dirty? Absolutely. Violating GPL? I think yes. Information about who is delivering main value has to be present in all context, not just in bad.

With Armbian you can make your own distribution in no time. With same "working good" or "working bad". Distro name is just a parameter:
https://docs.armbian.com/Developer-Guide_Build-Preparation/ There are many options for branding, etc.

> which seems to work fine,

Thanks to everyone but ...

> In any case I've found a very minimalistic distribution called DietPi 

Preinstalled with bunch of bloatware and minimal looks like oxymoron. Everything is made just to impress, which is no problem. htop is patched to adjust how numbers are displayed ... not so sure anymore. Why would you want to run heavily changed and with added security issues, for no usable reason, Debian like OS? 99.99% of changes are work of one person, were never reviewed, while Debian and Armbian is pretty much well reviewed and made by many. Dietpi also called gathered your data by default. Now I think only when you run update ;)  

[Propaganda](https://dietpi.com/stats.html). "Distributions that creates real value are trash". Comparison is fake but exists on prime entry location for many many years. I am sure a few noticed this too, but general population have no knowledge or reasons to not believe into this. But its just a fat lie and they are well aware of that.

DiegoBM, 2024-08-29:
Hi @igorpecovnik,

I can see that you are the armbian maintainer, so first of all thank you for your astounding work!

It seems like you have beef with the DietPi maintainer, which I won't discuss here since I don't know enough to talk, but from what little I have managed to research online so far, people seemed to be decently happy with their DietPi servers, that have been running for years with minimal maintenance. I didn't find any major security concerns but that might just mean that I didn't look hard enough, so I'll investigate a further since that's a critical topic. Your comments sound a bit biased so I'll take the most passionate parts with a grain of salt. Not saying that they are not true, but in my experience they are usually exagerated (I might be wrong), specially since you didn't provide any links to support those claims, in any case I'll try to investigate a bit further on what you mention.

Finally I'd love to use Joshua's distro or Armbian before any other options, unfortunately the latest releases for both is not working on my device, probably for the reason you mentioned and what we can read in the previous forum. I understand that armbian[ requires of funding to maintain a specific board](https://github.com/armbian/build/issues/6967#issuecomment-2238068527), which is fair, and I presume that for this distro, in general a board will get attention based on demand (I might be wrong). Unfortunately DietPi seems to be the only distro working on my card on it's latest version, so it's either that or Orange Pi's own flavour of Ubuntu server which I'm still not sure why people hate so much, I still haven't found any links with evidence supporting that it's spying on users, but it seems like it resets the root password on updates which is a bit weird... 

In any case thank you for that armbian build preparation link. I'm definitely interested in learningn how to build/maintain linux distros! And thank you again for your efforts (as well as @Joshua-Riek's)! Rest assure they are appreciated! :)

tomyjack, 2024-09-08:
我找到了解决办法：用PartitionGuru或者其他可以打开虚拟硬盘镜像的工具，把下载的Img文件打开，然后找一个官方镜像，找到rk3566-orangepi-3b-v2.dtb，官方镜像是放在专门的opi_boot分区里面的，把这个文件放到Img对应目录/usr/lib/firmware/5.10.0-1012-rockchip/device-tree/rockchip里面，然后重新用工具刷入就可以看到正常的安装系统画面了，原理是另外一个帖子里面发现没找到dtb所以就给它放了个dtb。至于系统版本的问题，有可能不是1012，大家自行修改。

DiegoBM, 2024-09-08:
Thank you @tomyjack. I'm not sure translate did the best job with your message but I reckon I got the important parts. I'll try your solution

tomyjack, 2024-09-14:
you can use like partitionguru tools to open the office image, then find rk3566-orangepi-3b-v2.dtb，put the file to this git release image，the path is "/usr/lib/firmware/5.10.0-1012-rockchip/device-tree/rockchip", and flash the image to the tf card test。i just test the 2204 desktop，it boot ok。but the gpu is like not work, so the desktop is slow.

DiegoBM, 2024-09-14:
@tomyjack thank you for taking the time to translate the message :). I presume there must be a way to fix the GPU, I was not aware of it not working since I'm aiming to use it headless

tonyzzp, 2024-09-20:
it's the corrent answer. the system boot successfully.

but i have another problem. the nvme disk doesn't display.

i tried 'lsblk'  'blkid'  'fdisk -l' 'lshw'. there is no nvme device.

> you can use like partitionguru tools to open the office image, then find rk3566-orangepi-3b-v2.dtb，put the file to this git release image，the path is [...]

DiegoBM, 2024-09-20:
According to https://github.com/Joshua-Riek/ubuntu-rockchip/issues/965#issuecomment-2336861782, Armbian is adding support for the 2.1 board

Gi-0n, 2024-09-21:
They already did it here is a debian image (minimal/iot) with support for orange pi 2.1 https://mega.nz/file/s9k3DaSK#_MPJY_7zdiSun5VwIC4Sk0lPcnQTz9unox91Rk3bUuo

closed 2025-02-10 by github-actions[bot]