Joshua-Riek/ubuntu-rockchip #985: Bug Report: Orange Pi 5 Plus - Ubuntu 24.04 server v2.3.0 doesn't boot

andrej-kutny, 2024-08-11:
### What happened?

I am trying to flash Ubuntu v2.3.0 to eMMC module but with no luck. I tried the same steps with v2.2.1 and it worked. Here are steps I took:

```
  sudo ubuntu-rockchip-install /dev/mmcblk0
sudo u-boot-install-mtd
```

 With version v2.2.1 it worked well. RIght after that I wanted to do the same with eMMC to reproduce the error but my device did not recognise eMMC - mmcblk0 was simply not there nor when I used `lsblk` command nor `sudo fdisk -l`. I have tried the same steps before when it recognised eMMC card and it just did not boot from eMMC. It is strange because if I poweroff the device, remove SD card it will boot from eMMC (v2.2.1) but v2.3.0 can't recognise eMMC / boot from it.

### Kernel version

6.1.0-1021-rockchip

### SBC model

Orange Pi 5 Plus

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

andrej-kutny, 2024-08-11:
So I have tried to flash v2.2.1 to SD Card and use dd command to flash v2.3.0 from v2.2.1: `xz -dc ubuntu-24.04-preinstalled-server-arm64-orangepi-5-plus.img.xz | sudo dd bs=4k of=/dev/mmcblk0` . After it finished I powered off the device, removed sd card and booted again and it booted from eMMC. So I have tried to `sudo u-boot-install-mtd` from v2.3.0 and it did not boot again.

So after that I tried to insert SD card with v2.2.1 again, call `sudo u-boot-install-mtd`, poweroff, remove sd card and boot again and v2.3.0 that is on eMMC has booted! So I think there might be some issue with new u-boot not recognising eMMC?

Joshua-Riek, 2024-08-11:
I'm aware of this issue and it has been fixed, the eMMC was not able to be found when booting from the SPI NOR. 

See: https://github.com/Joshua-Riek/ubuntu-rockchip/commit/50693e8acd8a3e7eabd386f97f6d6edfc4d9d989

andrej-kutny, 2024-08-11:
Perfect, thank you for quick reply and I am happy that you fixed it already :) I am closing the issue.

closed 2024-08-11 by andrej-kutny