Joshua-Riek/ubuntu-rockchip #987: Feature Request: Port RSetup To This Build?

Jibun-no-Kage, 2024-08-14:
### Feature Description

RSetup is key to maintaining the hardware side of the ZERO 3 models of the RADXA SBCs, such as maintaining the bootloader.

Joshua-Riek, 2024-08-14:
This tool is specific to Radxa's custom operating systems. I'm trying to make this version of Ubuntu as close as I can to upstream without any hacks. 

I would consider the rsetup utility a hack. If you can provide detailed information on the specific functionality you are looking for, there may be an alternative solution.

closed 2024-08-14 by Joshua-Riek

Jibun-no-Kage, 2024-08-17:
Yeah, I see your point.  Rsetup provides one key feature in my book... the ability to change the bootloader configuration and update its code.  But I would call that and edge case, not all users will know or care about it?

hgpuke, 2024-08-24:
There is a Github repo for rsetup here: https://github.com/radxa-pkg/rsetup
It has a makefile that can build a Debian package. That should work inside your Ubuntu-version, should it not?

Joshua-Riek, 2024-08-24:
Just because it's a debian package does not mean it will work. Radxa has their system setup in a specific way to handle the boot process and manage updates. I do both in a completely different way.

hgpuke, 2024-08-25:
Understood. But does it meant that _nothing_ in rsetup works, or that _some things_ does not work?

Joshua-Riek, 2024-08-25:
The fan works, I do not see how rsetup affects the fan.

hgpuke, 2024-08-25:
I haven't received my fan yet, so I am not sure, but what I can read from Radxas document, the rsetup utility allows you to choose a "thermal governor" (which I believe is the same as a policy, so that the fan acts according to your preference.

Jibun-no-Kage, 2024-08-26:
Yeah... about the RADXA customization of the boot sequence.  I discovered that apparently, u-boot-update completely ignores /etc/default/u-boot, so any customization done in /boot/extlinux/extlinux.conf is wiped out... that is understood.  But the way to customize u-boot is per u-boot documentation via /etc/default/u-boot, so I did that... for example I added ipv6.disable=1 to the kernel parameters list... it was completely ignored when I ran u-boot-update.  This from my perspective means that RADXA broke the u-boot customization feature set.  I am pretty sure I am reading the u-boot documentation correctly and it is clear that per same documentation how it should behave.

hgpuke, 2024-08-26:
Sorry to hear that. By the way, I have tried a lot of Linux distros on different hardware in the past, and your customization of Ubuntu for RK3588 is by far the best experience I have had with a SBC. So congratulations!