Joshua-Riek/ubuntu-rockchip #988: Bug Report: Unable to boot from NVME on Radxa 5b+

stsvilik, 2024-08-14:
### What happened?

1. I have successfully installed Ubuntu 24.04 LTS onto an SD card
2. Ran command to install U-Boot to SPI Flash (with success)
3. Ran command to install Ubunto onto NVME (with success)

When rebooting board w/out SD card inserted, a blue status light stays solid and nothing is shown on display.
System detects INTEL SSD just fine and shows it in both `fdisk` and `lsblk`

### Kernel version

5.10.0-1011-rockchip

### SBC model

Radxa 5B+

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

wdschn, 2024-08-14:
I have the same problem.

Joshua-Riek, 2024-08-14:
I recall U-Boot did not support proper PCIe Bifurcation when I was originally working on this device. 

Needs some research to determine if this is still the case.

stsvilik, 2024-08-14:
@Joshua-Riek let me know if you need me to run/log/inspect anything. Thank you!

Joshua-Riek, 2024-08-14:
All the info needed will be in the bootloader. If you can connect a uart cable and post the boot logs that would be helpfull.

stsvilik, 2024-08-15:
Ordered [this](https://www.amazon.com/dp/B07S7YSGVV?psc=1&smid=A2ZDGCOOU4F0SF&ref_=chk_typ_imgToDp). Will get back once I get a chance to connect.

pascal71, 2024-08-16:
Booting from NVME works for me; even with 2 NVMEs

ferrellsl, 2024-08-16:
I had a similar problem on the Orange Pi 5.  What worked for me was to boot with the SD card.  Then I opened a terminal and: sudo apt update && upgrade, then rebooted once the OS on the SD card had been updated.  Next, I wrote the boot loader to the SPI:  sudo u-boot-install-mtd

Then I wrote the contents of the SD card to the nvme:  ubuntu@ubuntu:~$ sudo ubuntu-rockchip-install /dev/nvme0n1

Installation wouldn't stick until I had performed the update AND the upgrade first for some reason.  I tried installing without updating 2-3 times and it would always give me a blank screen and the power light would stay on solid and that was it.

I was upgrading from one of Joshua's releases from over a year ago so that may have been a factor too.

Joshua-Riek, 2024-08-16:
> I had a similar problem on the Orange Pi 5. What worked for me was to boot with the SD card. Then I opened a terminal and: sudo apt update && upgrade, then rebooted once the OS on the SD card had [...]

This is a very different device with a different problem, the Orange Pi 5 had a SPL issue.

stsvilik, 2024-08-16:
@Joshua-Riek I tried getting SPI logs today using https://docs.radxa.com/en/nio/nio12l/low-level-dev/serial?os=mac but output is unreadable, do you know if they've changed boud rate on serial?

Joshua-Riek, 2024-08-16:
Set baud rate to 1500000, if it does not work swap rx and tx.

stsvilik, 2024-08-16:
> Set baud rate to 1500000, if it does not work swap rx and tx.

Still see giberish - I'll order a different cable, maybe the one I ordered has boud rate limitations.

stsvilik, 2024-08-17:
@Joshua-Riek here is the output

```
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR5, 2400MHz
channel[0] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[1] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[2] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[3] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
Manufacturer ID:0x6
CH0 RX Vref:27.9%, TX Vref:23.0%,23.0%
CH1 RX Vref:27.5%, TX Vref:24.0%,25.0%
CH2 RX Vref:29.7%, TX Vref:21.0%,21.0%
CH3 RX Vref:30.1%, TX Vref:19.0%,20.0%
change to F1: 534MHz
change to F2: 1320MHz
change to F3: 1968MHz
change to F0: 2400MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09 (Aug 09 2024 - 03:21:33)
Trying to boot from MMC2
spl: partition error
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(a7d1d8d191...) + OK
## Checking uboot 0x00200000 ... sha256(9eab49d292...) + OK
## Checking fdt 0x0031fb50 ... sha256(38228f5acd...) + OK
## Checking atf-2 0xff100000 ... sha256(4b2065349b...) + OK
## Checking atf-3 0x000f0000 ... sha256(aa71013e72...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 796.804/1106.976 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-682-g4ca8a8422:derrick.huang, fwver: v1.45
NOTICE:  BL31: Built : 10:11:21, Dec 27 2023
INFO:    spec: 0x1
INFO:    code: 0x88
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    l3 cache partition cfg-0
INFO:    system boots from cpu-hwid-0
INFO:    disable memory repair
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2400MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 534MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1320MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1968MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2017.09 (Jul 30 2024 - 00:26:32 +0000)

Model: Radxa ROCK 5B+
MPIDR: 0x81000000
Praw, 0xfeb50000
DRAM:  16 GiB
Sysmem: init
Relocation Offset: eda3e000
Relocation fdt: eb9f8820 - eb9fecd0
CR: M/C/I
Using default environment
[... 355 lines cut ...]
[   30.112160] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.112270] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.112363] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.112471] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.113563] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.113618] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.113665] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.113706] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.113762] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114018] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114066] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114117] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114158] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114212] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114562] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114659] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114753] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114832] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.114926] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.115353] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.115463] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.115570] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.115664] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   30.115773] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
```

stsvilik, 2024-08-17:
> > Set baud rate to 1500000, if it does not work swap rx and tx.
> 
> Still see giberish - I'll order a different cable, maybe the one I ordered has boud rate limitations.

That was it! New one is based on CH340 and it works flawlessly.

Joshua-Riek, 2024-08-19:
Ah, I need to update the u-boot package to the latest upstream from Radxa. The PCIe nodes are currently disabled during the boot process, preventing any NVMe drives from being detected.

Joshua-Riek, 2024-08-22:
Package has been updated. Users can update their system with apt, then update u-boot with `sudo u-boot-install-mtd`.

stsvilik, 2024-08-22:
All works now! Thank you so much @Joshua-Riek!

P.S. If you need any additional boot logs please let me know.