Joshua-Riek/ubuntu-rockchip #992: Bug Report: Blank Screen after Installation on Orange Pi 5

ferrellsl, 2024-08-15:
### What happened?

I tried installing using both methods below:

xz -dc ubuntu-22.04-preinstalled-desktop-arm64-orange-pi5.img.xz | sudo dd of=/dev/nvme0n1 bs=4k

ubuntu@ubuntu:~$ sudo ubuntu-rockchip-install /dev/nvme0n1

I've also made sure to run the following:  sudo u-boot-install-mtd

The system WILL boot from the SD card, but after performing the above, and performing a cold start, all I get is a blank screen and the red power light with no other activity whatsoever.  Am I missing something?

### Kernel version

5.10.0.1011-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

### Relevant logs

```shell
No relevant logs available.
```

Joshua-Riek, 2024-08-15:
Update the system first. I cant find the issue number but there was a problem with the SPL in the bootloader.

ferrellsl, 2024-08-15:
Never mind.  I installed the image again and now all is working now.  I guess something went wrong the first time around.

closed 2024-08-15 by ferrellsl