Joshua-Riek/ubuntu-rockchip #993: Bug Report: AIO 3588L - Just cursor and wallpaper after installation

spedagadi, 2024-08-16:
### What happened?

Hi 

I have installed  version 24.04 https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.3.0/ubuntu-24.04-preinstalled-desktop-arm64-aio-3588l.img.xz on my AIO 3588L and soon after installation, it will not show full desktop, rather only the cursor and wallpaper. I can still open terminal.

When I enter console model with ctl+alt+f3, a series of errors get printed as shown. 

[image]

Could I be doing any thing wrong during installation? The images seemed to have worked on orange pi 5 plus and rock ITX boards. 

thnx

### Kernel version

6.1

### SBC model

AIO 3588L

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
attached as screenshot
```

Joshua-Riek, 2024-08-16:
I don't have this board, and the device tree is kinda messy. If you can post the full boot logs from dmesg I can take a look. 

Also, you may have a bad install.

spedagadi, 2024-08-27:
hi @serwansj you mention in https://github.com/Joshua-Riek/ubuntu-rockchip/issues/930 that AIO 3588L is now supported. Wonder if you are able to share the link to the prebuilt image that worked for you as I am having issue as described above with the latest image? thnx

serwansj, 2024-08-27:
Hi @spedagadi, I have not tested the desktop image, but the server image should be working

closed 2025-02-10 by github-actions[bot]