Joshua-Riek/ubuntu-rockchip #995: Bug Report:  the pc can't start after  I modified the u-boot file

augetyz, 2024-08-18:
### What happened?

```
## /etc/default/u-boot - configuration file for u-boot-update(8)

#U_BOOT_UPDATE="true"

#U_BOOT_ALTERNATIVES="default recovery"
#U_BOOT_DEFAULT="l0"
#U_BOOT_PROMPT="1"
#U_BOOT_ENTRIES="all"
#U_BOOT_MENU_LABEL="Debian GNU/Linux"
#U_BOOT_PARAMETERS="ro earlycon"
#U_BOOT_ROOT=""
#U_BOOT_TIMEOUT="50"
U_BOOT_FDT="device-tree/rockchip/rk3588s-rock-5a.dtb"
U_BOOT_FDT_DIR="/lib/firmware/lib/firmware/5.10.0-1011-rockchip/"
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-pwm14-m2.dtbo device-tree/rockchip/overlay/rk3588-pwm6-m0.dtbo device-tree/rockchip/overlay/rk3588-uart4-m2.dtbo device-tree/rockchip/overlay/rk3588-uart7-m2.dtbo  device-tree/rockchip/overlay/rock-5a-radxa-camera-8m-219.dtbo device-tree/rockchip/overlay/rk3588-uart2-m0.dtbo  device-tree/rockchip/overlay/rk3588-i2c8-m2.dtbo device-tree/rockchip/overlay/rk3588-spi0-m1-cs1-spidev.dtbo device-tree/rockchip/overlay/rk3588-can1-m1.dtbo device-tree/rockchip/overlay/rk3588-spi4-m2-cs0-spidev.dtbo"
#U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/"
#U_BOOT_SYNC_DTBS="false"
```

### Kernel version

ubuntu22.04

### SBC model

rock 5a

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

### Relevant logs

```shell
the pc can't start after  I modified the u-boot file,The main parameter is U_BOOT_FDT_DIR
```

augetyz, 2024-08-18:
how can i recover the last status，i tried Paragon ExtFS for Windows software,but no effect

wmport, 2024-08-18:
sudo u-boot-update
after  modified the u-boot file

augetyz, 2024-08-18:
I have modified input upadte order.
And now I can’t start pc after reboot .I think the cause may be I added the option `U_BOOT_FDT="device-tree/rockchip/rk3588s-rock-5a.dtb"`
Now I can’t start 😭😭

moo-the-cow, 2024-08-18:
what I sometimes do because I made a silly typo once in a while just mount the drive on another linux (live boot image also works fine for that)  and remove/modify the entry from /boot/extlinux/extlinux.conf

closed 2024-08-19 by augetyz