Joshua-Riek/ubuntu-rockchip #1003: Bug Report: Latest Update Makes Wifi Dissapear

AdimasBarelang, 2024-08-23:
### What happened?

I updated my Orange Pi 5 Pro using the command "sudo apt-get update && sudo apt-get upgrade". After that, to ensure everything was fully updated, I used the Ubuntu Software Updater. The updater required a restart, and after rebooting, I noticed a problem: my Wi-Fi adapter stopped working, and the Wi-Fi icon was missing.

### Kernel version

6.1.0-1023-rockchip

### SBC model

Orange Pi 5 Pro

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
<script src="https://gist.github.com/AdimasBarelang/70dc601bdacc505e3ce7db988698b2d2.js"></script>
```

Joshua-Riek, 2024-08-23:
```bash

``` 
nstall panfork
        chroot "${rootfs}" add-apt-repository -y ppa:jjriek/panfork-mesa
        chroot "${rootfs}" apt-get update
        chroot "${rootfs}" apt-get -y install mali-g610-firmware
        chroot "${rootfs}" apt-get -y dist-upgrade

        # Install libmali blobs alongside panfork
        chroot "${rootfs}" apt-get -y install libmali-g610-x11

        # Install the rockchip camera engine
        chroot "${rootfs}" apt-get -y install camera-engine-rkaiq-rk3588

        # Install BCMDHD SDIO WiFi and Bluetooth DKMS
        chroot "${rootfs}" apt-get -y install dkms bcmdhd-sdio-dkms

Joshua-Riek, 2024-08-23:
This WiFi driver has caused me a lot of problems, try and run this command then reboot your system.
```bash
echo 'mmc2:0001:1' | sudo tee /sys/bus/sdio/drivers/brcmfmac/unbind
```
See https://github.com/Joshua-Riek/ubuntu-rockchip/issues/841#issuecomment-2249455427

Joshua-Riek, 2024-08-23:
If that does not work you can try to install the dkms package.
```bash
echo "blacklist bcmdhd" | sudo tee -a /etc/modprobe.d/bcmdhd.conf
echo "blacklist dhd_static_buf" | sudo tee -a /etc/modprobe.d/bcmdhd.conf
sudo apt-get update
sudo apt-get -y install dkms bcmdhd-sdio-dkms
```

AdimasBarelang, 2024-08-29:
I have resolved this issue by simply unplugging the power source and turning it back on. After that, the board wifi ran normally again. I haven't tried your solution yet, but I will close this issue. Thank you for your help and feedback.

matteoguglielmi, 2024-09-04:
@Joshua-Riek Hi Joshua, I have just installed your latest image 2.3.1 for my OrangePI 5 Pro and faced exactly the same wifi issue after the first reboot. The first solution (`echo 'mmc2:0001:1' > /sys/bus/sdio/drivers/brcmfmac/unbind`) does not work for me since I get this error meaasege `-bash: echo: write error: No such device` and the second solution does not make any difference since the wifi keeps disappearing after the first reboot.

These are the modules being loaded after applying solution number two:

```
[root@mlab:~]# lsmod | grep bcmdhd
bcmdhd_sdio          1449984  0
dhd_static_buf_sdio    16384  1 bcmdhd_sdio
```

Thank you for any help.