Joshua-Riek/ubuntu-rockchip #1004: Bug Report:

ArtemS2, 2024-08-24:
### What happened?

Cant use ping
ping 8.8.8.8
ping: socktype: SOCK_RAW
ping: socket: Операция не позволена
ping: => missing cap_net_raw+p capability or setuid?

### Kernel version

6.1.0-1023-rockchip

### SBC model

Orange PI5B

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
ping 8.8.8.8
ping: socktype: SOCK_RAW
ping: socket: Операция не позволена
ping: => missing cap_net_raw+p capability or setuid?
```

hanvinke, 2024-08-24:
Same here with nano-pc T6

`ping 8.8.8.8
ping: socktype: SOCK_RAW
ping: socket: Operation not permitted
ping: => missing cap_net_raw+p capability or setuid?`

but: 

`sudo ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=116 time=19.6 ms`

Joshua-Riek, 2024-08-24:
I highly doubt this is a problem with Ubuntu Rockchip, check your ISP and internet settings.

ArtemS2, 2024-08-24:
@Joshua-Riek If the problem was with the connection, then the ping from the sudo user also did not pass.

hanvinke, 2024-08-24:
sudo sysctl -w net.ipv4.ping_group_range="0 1000"

han@han-desktop:~$ sudo ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=116 time=21.2 ms

ArtemS2, 2024-08-24:
@hanvinke  Thank you very much this really helped.

closed 2025-02-10 by github-actions[bot]