Joshua-Riek/ubuntu-rockchip #1006: Bug Report: Mini Jack output not work NanoPC T6

Xeue, 2024-08-25:
### What happened?

No audio on the stereo minijack output.
Issue appears to be in armbian kernel, same as this issue:
https://github.com/armbian/linux-rockchip/issues/219

Following in kernel logs:
`[18524.226883] asoc-simple-card rt5616-sound: ASoC: DAPM unknown pin Headphones`

I was testing in gstreamer using the folowing command:
`gst-launch-1.0 audiotestsrc ! audioconvert ! audioresample ! alsasink device=hw:4`

Your audio device may be a different number `aplay -l` will list the devices

### Kernel version

6.1.0-1014-rockchip

### SBC model

FriendlyElec NanoPC-T6 LTS

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
[18524.226883] asoc-simple-card rt5616-sound: ASoC: DAPM unknown pin Headphones
```

Tiemichael, 2024-10-13:
Yes, I have the same entry in my dmesg (but no idea what's the relevance:
` dmesg | grep sound
[    9.547416] input: rockchip-hdmiin rockchip-hdmiin as /devices/platform/hdmiin-sound/sound/card0/input2
[    9.923162] asoc-simple-card rt5616-sound: ASoC: DAPM unknown pin Headphones
[    9.926056] input: realtek,rt5616-codec Headphones as /devices/platform/rt5616-sound/sound/card1/input3
[    9.929264] input: rockchip-dp0 rockchip-dp0 as /devices/platform/dp0-sound/sound/card3/input4
[   10.077055] asoc-simple-card rt5616-sound: ASoC: DAPM unknown pin Headphones
`

Tiemichael, 2024-10-13:
Yes, same dmesg, but no idea if it's relevant ....

closed 2025-02-10 by github-actions[bot]