Joshua-Riek/ubuntu-rockchip #1007: Bug Report: no wifi or bluetooth with Indiedroid Nova 24.04 image

alansrobotlab2, 2024-08-26:
### What happened?

Onboard wifi settings aren't available on boot or after sudo apt upgrade.  Bluetooth settings are visible in ui but can't turn on.  In addition my usb keyboard adapter isn't recognized until I remove and replace it after booting in case it provides any hints.

Thanks for your help!

### Kernel version

6.1.0-1024-rockchip

### SBC model

Indiedroid Nova

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
alansrobotlab@rufus:~$ lspci
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller (rev 15)

dmesg segments

[    9.734669] kernel: NET: Registered PF_PACKET protocol family
[    9.734946] kernel: [BT_RFKILL]: Enter rfkill_rk_init
[    9.734949] kernel: [WLAN_RFKILL]: Enter rfkill_wlan_init
[    9.735338] kernel: [WLAN_RFKILL]: Enter rfkill_wlan_probe
[    9.735360] kernel: [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = rtl8821cs
[    9.735362] kernel: [WLAN_RFKILL]: wlan_platdata_parse_dt: enable wifi power control.
[    9.735364] kernel: [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[    9.735387] kernel: [WLAN_RFKILL]: wlan_platdata_parse_dt: WIFI,host_wake_irq = 0, flags = 0.
[    9.735392] kernel: [WLAN_RFKILL]: wlan_platdata_parse_dt: The ref_wifi_clk not found !
[    9.735394] kernel: [WLAN_RFKILL]: rfkill_wlan_probe: init gpio
[    9.735396] kernel: [WLAN_RFKILL]: rfkill_set_wifi_bt_power: 1
[    9.735398] kernel: [WLAN_RFKILL]: Exit rfkill_wlan_probe
[    9.735694] kernel: [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: uart_rts_gpios = 122.
[    9.735702] kernel: [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,reset_gpio = 28.
[    9.735706] kernel: [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_gpio = 22.
[    9.735710] kernel: [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_host_irq = 21.
[    9.735744] kernel: [BT_RFKILL]: Request irq for bt wakeup host
[    9.736014] kernel:  mmcblk0: p1
[    9.736094] kernel: [BT_RFKILL]: ** disable irq
[    9.736148] kernel: [BT_RFKILL]: bt_default device registered.

[   10.414504] kernel: r8169 0004:41:00.0: enabling device (0000 -> 0003)
[   10.428691] kernel: r8169 0004:41:00.0: can't read MAC address, setting random one
[   10.465080] kernel: r8169 0004:41:00.0 eth0: RTL8168h/8111h, 52:1d:1f:9d:03:45, XID 541, IRQ 134
[   10.465101] kernel: r8169 0004:41:00.0 eth0: jumbo features [frames: 9194 bytes, tx checksumming: ko]
[   10.466476] kernel: r8169 0004:41:00.0 enP4p65s0: renamed from eth0

[   12.291279] kernel: rtw_8821cs mmc2:0001:1: Firmware version 24.11.0, H2C version 12
[   12.297973] kernel: [BT_RFKILL]: rfkill_rk_set_power: set bt wake_host high!
[   12.353380] kernel: [BT_RFKILL]: rfkill_rk_set_power: set bt wake_host input!
[   12.353406] kernel: [BT_RFKILL]: ENABLE UART_RTS
[   12.460015] kernel: [BT_RFKILL]: DISABLE UART_RTS
[   12.460060] kernel: [BT_RFKILL]: bt turn on power
[   12.460211] kernel: [BT_RFKILL]: Request irq for bt wakeup host
[   12.460592] kernel: [BT_RFKILL]: ** disable irq
[   12.625912] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x11080): -110
[   12.626007] kernel: rtw_8821cs mmc2:0001:1: sdio write32 failed (0x11080): -110
[   12.626247] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x10080): -110
[   12.626327] kernel: rtw_8821cs mmc2:0001:1: sdio write32 failed (0x10080): -110
[   12.626406] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x10040): -110
[   12.626487] kernel: rtw_8821cs mmc2:0001:1: sdio write32 failed (0x10040): -110
[   12.626643] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x11700): -110
[   12.626745] kernel: rtw_8821cs mmc2:0001:1: sdio write32 failed (0x11700): -110
[   12.626847] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x11708): -110
[   12.627155] kernel: rtw_8821cs mmc2:0001:1: sdio write32 failed (0x11330): -110
[   12.627303] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x1022c): -110
[   12.627450] kernel: rtw_8821cs mmc2:0001:1: sdio write32 failed (0x1022c): -110
[   12.627550] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x20): -110
[   12.627891] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x11208): -110
[   12.627942] kernel: rtw_8821cs mmc2:0001:1: sdio write32 failed (0x11208): -110
[   12.628128] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x20): -110
[   12.628180] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x24): -110
[   12.628232] kernel: rtw_8821cs mmc2:0001:1: sdio read32 failed (0x28): -110
[   12.628305] kernel: rtw_8821cs mmc2:0001:1: Failed to write 4608 byte(s) to SDIO port 0x0000840c
[   12.628317] kernel: rtw_8821cs mmc2:0001:1: failed to write data to rsvd page
[   12.628423] kernel: rtw_8821cs mmc2:0001:1: failed to download rsvd page
[   12.628522] kernel: rtw_8821cs mmc2:0001:1: failed to download firmware
[   12.628620] kernel: rtw_8821cs mmc2:0001:1: failed to setup chip efuse info
[   12.628633] kernel: rtw_8821cs mmc2:0001:1: failed to setup chip information
[   12.628867] kernel: rtw_8821cs: probe of mmc2:0001:1 failed with error -110
[   12.720602] kernel: audit: type=1400 audit(1724642536.500:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="snap-update-ns.lxd" pid=671 comm="apparmor_parse>
[   12.726647] kernel: audit: type=1400 audit(1724642536.506:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/snap/snapd/21761/usr/lib/snapd/snap-confine" pi>
[   12.726659] kernel: audit: type=1400 audit(1724642536.506:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/snap/snapd/21761/usr/lib/snapd/snap-confine//mo>
[   12.730989] kernel: audit: type=1400 audit(1724642536.510:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="snap.lxd.activate" pid=672 comm="apparmor_parser"
[   12.737458] kernel: audit: type=1400 audit(1724642536.516:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="snap.lxd.hook.configure" pid=676 comm="apparmor_>
[   12.738414] kernel: audit: type=1400 audit(1724642536.516:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="snap.lxd.daemon" pid=675 comm="apparmor_parser"
[   12.738821] kernel: audit: type=1400 audit(1724642536.516:8): apparmor="STATUS" operation="profile_load" profile="unconfined" name="snap.lxd.hook.connect-plug-ovn-chassis" pid=679 >
[   12.739039] kernel: audit: type=1400 audit(1724642536.516:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="snap.lxd.check-kernel" pid=674 comm="apparmor_pa>
[   12.739179] kernel: audit: type=1400 audit(1724642536.516:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="snap.lxd.hook.connect-plug-ceph-conf" pid=677 c>
[   12.954449] kernel: NET: Registered PF_QIPCRTR protocol family
[   13.418304] kernel: loop5: detected capacity change from 0 to 8
[   14.170103] kernel: Generic FE-GE Realtek PHY r8169-4-4100:00: attached PHY driver (mii_bus:phy_addr=r8169-4-4100:00, irq=MAC)
[   14.363500] kernel: r8169 0004:41:00.0 enP4p65s0: Link is Down
[   14.435750] kernel: [BT_RFKILL]: bt shut off power
[   16.359178] kernel: mali fb000000.gpu: Loading Mali firmware 0x1010000
[   16.360692] kernel: mali fb000000.gpu: Mali firmware git_sha: ee476db42870778306fa8d559a605a73f13e455c
[   16.464752] kernel: [BT_RFKILL]: rfkill_rk_set_power: set bt wake_host high!
[   16.516738] kernel: [BT_RFKILL]: rfkill_rk_set_power: set bt wake_host input!
[   16.516768] kernel: [BT_RFKILL]: ENABLE UART_RTS
[   16.623345] kernel: [BT_RFKILL]: DISABLE UART_RTS
[   16.623374] kernel: [BT_RFKILL]: bt turn on power
[   16.623435] kernel: [BT_RFKILL]: Request irq for bt wakeup host
[   16.623837] kernel: [BT_RFKILL]: ** disable irq
[   18.011920] kernel: r8169 0004:41:00.0 enP4p65s0: Link is Up - 1Gbps/Full - flow control off
[   18.011944] kernel: IPv6: ADDRCONF(NETDEV_CHANGE): enP4p65s0: link becomes ready
[   18.640067] kernel: Bluetooth: HCI UART driver ver 2.3
[   18.640074] kernel: Bluetooth: HCI UART protocol H4 registered
[   18.640076] kernel: Bluetooth: HCI UART protocol BCSP registered
[   18.640078] kernel: Bluetooth: HCI UART protocol ATH3K registered
[   18.640108] kernel: Bluetooth: HCI UART protocol Three-wire (H5) registered
[   18.640223] kernel: Bluetooth: HCI UART protocol Intel registered
[   18.640277] kernel: Bluetooth: HCI UART protocol Broadcom registered
[   18.640290] kernel: Bluetooth: HCI UART protocol QCA registered
[   18.640292] kernel: Bluetooth: HCI UART protocol AG6XX registered
```

numetaler, 2024-09-03:
Hello, has someone found a solution yet?

Jibun-no-Kage, 2024-09-18:
Anyone working on this?  Is there a work around?

Jibun-no-Kage, 2025-02-05:
I can confirm this behavior even in the new Debian 12 image that was released on 01/25/25.  The workaround is to unload the driver stack and reload AFTER boot.

```
# lsmod | grep rtw88
rtw88_8821cs           16384  0
rtw88_8821c            90112  1 rtw88_8821cs
rtw88_sdio             24576  1 rtw88_8821cs
rtw88_core            163840  2 rtw88_8821c,rtw88_sdio
# rmmod rtw88_8821cs
# rmmod rtw88_8821c
# rmmod rtw88_sdio
# rmmod rtw88_core
# lsmod | grep rtw88

# modprobe rtw88_core
# modprobe rtw88_8821cs
# lsmod | grep rtw88
rtw88_8821cs           16384  0
rtw88_8821c            90112  1 rtw88_8821cs
rtw88_sdio             24576  1 rtw88_8821cs
rtw88_core            163840  2 rtw88_8821c,rtw88_sdio
# ifconfig wlan0
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Jibun-no-Kage, 2025-03-31:
One would hope someone works in this issue?

jimdinunzio, 2025-05-18:
> I can confirm this behavior even in the new Debian 12 image that was released on 01/25/25. The workaround is to unload the driver stack and reload AFTER boot.
> 

Ok, that works for Ubuntu 24.04 Indiedroid Nova for Wifi, but BT still cannot be turned on. 
I tried several steps, but it will not load. same errors as above with prefix [BT_RFKILL]