Joshua-Riek/ubuntu-rockchip #1008: Bug Report:  one USB 3.0 becomes to USB 2.0

mario-huang, 2024-08-26:
### What happened?

There should be two USB 3.0 ports. 
[image]

Bus 004 should be USB 3.0.
$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 001: Dev 002, If 0, Class=Mass Storage, Driver=uas, 480M
/:  Bus 005.Port 001: Dev 001, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 006.Port 001: Dev 001, Class=root_hub, Driver=ohci-platform/1p, 12M

### Kernel version

6.1.0-1023-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Joshua-Riek, 2024-08-26:
There is only one USB 3.0 port. The rest are USB 2.0.

mario-huang, 2024-08-26:
> There is only one USB 3.0 port. The rest are USB 2.0.

What? I see there are two in the document, one is 3.0, one is 3.1. Am I wrong?
[image]

Joshua-Riek, 2024-08-26:
The 3.1 USB port is shared between USB-C and the white USB 2.0 port.

mario-huang, 2024-08-26:
$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 005.Port 001: Dev 001, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 006.Port 001: Dev 001, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 007.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 008.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 5000M

mario-huang, 2024-08-26:
> The 3.1 USB port is shared between USB-C and the white USB 2.0 port.

Yes, that's true, but I think there are two blue USB  ports. The up one is 3.0 and the down one is 2.0.
Sorry for the newbie questions.😀

Joshua-Riek, 2024-08-26:
I see nothing wrong here, as stated previously. If you believe there is, you are welcome to investigate.

closed 2024-08-26 by Joshua-Riek