Joshua-Riek/ubuntu-rockchip #1011: Bug Report:  I lost my ethernet nics on kernel 6.1.0-1024-rockchip

YuanLeeping, 2024-08-26:
### What happened?

**$0 Kernel 6.1.0-1023-rockchip**
ubuntu@orangepi-5plus:~$ sudo lspci -Dk
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
	Kernel driver in use: pcieport
0000:01:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO
	Subsystem: Samsung Electronics Co Ltd SSD 980 PRO
	Kernel driver in use: nvme
0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
	Kernel driver in use: pcieport
0003:31:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
	Subsystem: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller
	Kernel driver in use: r8169
	Kernel modules: r8169
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
	Kernel driver in use: pcieport
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
	Subsystem: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller
	Kernel driver in use: r8169
	Kernel modules: r8169

**$1 Kernel Kernel 6.1.0-1024-rockchip**
ubuntu@orangepi-5plus:~$ sudo lspci -Dk
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
	Kernel driver in use: pcieport
0000:01:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO
	Subsystem: Samsung Electronics Co Ltd SSD 980 PRO
	Kernel driver in use: nvme
0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
	Kernel driver in use: pcieport
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
	Kernel driver in use: pcieport

If you want to more info, please tell me.  Thank you.
[k23_bootlog.txt](https://github.com/user-attachments/files/16750743/k23_bootlog.txt)
[k24_bootlog.txt](https://github.com/user-attachments/files/16750745/k24_bootlog.txt)

### Kernel version

6.1.0-1024-rockchip

### SBC model

orangepi-5plus + SSD boot 

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

YuanLeeping, 2024-08-27:
I test **6.1.0-1025-rockchip**. It works well. Thank you very much!

stevepringle, 2024-08-28:
> Thank you for the detailed logs and bug report.
> This commit is the problem ***@***.*** It was meant to fix SATA hard drives on the Rock 5 ITX, I should have done more testing on other hardware before releasing the Ubuntu-rockchip-6.1.0-1024.24 kernel package. I'm very sorry about the inconvenience.
> I will revert this change tonight and push a new kernel update.

Hi Joshua,Any idea when the Rock 5 itx SATA issue is likely to get fixed now?Regards,SteveSent from my iPad

Joshua-Riek, 2024-08-28:
I have no Idea about the Rock 5 ITX SATA issue.

closed 2024-08-28 by Joshua-Riek