Joshua-Riek/ubuntu-rockchip #1012: Bug Report: Orange Pi 5 - Unable to boot with NVMe after upgrade to kernel 6.1.0-1024-rockchip

lucianor, 2024-08-26:
### What happened?

After upgrading to kernel 6.1.0-1024-rockchip, I was unable to boot from the NVMe. Had to connect the HDMI on my Orange Pi and then check the kernel logs - it was booting to initramfs.

It looks like it fails to probe for the NVMe drive, and then just fail. I took a picture of the monitor - I can try to capture the logs from a UART but it will take me longer. 

[image]

Reverting back to kernel 6.1.0-1023-rockchip fixed the problem. For those with the same problem, it's an easy fix: write one of the images for Orange Pi 5, and then mount and chroot to the NVMe drive:

mkdir /rescue
mount /dev/nvme0n1p2 /rescue
mount -t proc proc /rescue/proc
mount -t sysfs sys /rescue/sys
mount -o bind /dev /rescue/dev
mount -t devpts pts /rescue/dev/pts
chroot /rescue
apt purge linux-image-6.1.0-1024-rockchip
update-initramfs -u -k all
exit
reboot

### Kernel version

6.1.0-1024-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Bluesmanuk, 2024-08-26:
I can confirm having the same issue with my Orange Pi 5.

Joshua-Riek, 2024-08-26:
Currently waiting on the kernel package to finish building as a fix.

https://launchpad.net/~jjriek/+archive/ubuntu/rockchip/+build/28896374

JanV123, 2024-08-27:
noob question: how do I "write one of the images for Orange Pi 5"? I did a `>ls /dev/` onto the terminal with the initrams prompt and I don't see my nvme (/dev/nvme0n1p2) drive there.

lucianor, 2024-08-27:
You need another computer to write one of the images for Orange Pi 5 - you don`t do it from the initramfs prompt.
1) Download an image from the [release page ](https://github.com/Joshua-Riek/ubuntu-rockchip/releases) 2) 
2) Use [balena Etcher](https://etcher.balena.io/#download-etcher) or your tool of choice to write that image to the SD card
3) Insert the SD card into OPI 5 and boot from it, and then follow the instructions on the bug report.

JanV123, 2024-08-27:
got it. Thanks!

Esteban5713, 2024-08-27:
I confirm that I have the same problem, I updated the system and it no longer started and I lost all my files that I had 😅

Joshua-Riek, 2024-08-27:
> I confirm that I have the same problem, I updated the system and it no longer started and I lost all my files that I had 😅

You can connect a serial console and select the previous kernel version, or modify /boot/extlinux/extlinux.conf manually to select the previous kernel.

Esteban5713, 2024-08-28:
Thank you, that will no longer be possible because when I saw that it did not start, I formatted it and reinstalled it again, I thought I had done something wrong, now I will wait for the solution and thank you very much for the good work.

JanV123, 2024-08-28:
> You need another computer to write one of the images for Orange Pi 5 - you don`t do it from the initramfs prompt.
> 
>     1. Download an image from the [release page ](https://github.com/Joshua-Riek/ubuntu-rockchip/releases)
> 
>     2. Use [balena Etcher](https://etcher.balena.io/#download-etcher) or your tool of choice to write that image to a SD card
> 
>     3. Insert the SD card into OPI 5 and boot from it, and then follow the instructions on the bug report.

UPDATE: got my OP5 working again. Thanks for your help mate.

closed 2024-08-28 by Joshua-Riek

lucianor, 2024-08-28:
> UPDATE: got my OP5 working again. Thanks for your help mate. That said, I did a `sudo apt-get update && sudo apt-get dist-upgrade` and it found the new linux-image-6.1.0-1025-rockchip but it proceeded to stick with the linux-image-6.1.0-1023-rockchip. It downloaded the former but just deleted it afterwords. Is there a config that needs to be updated to resolve this?

`sudo apt install linux-image-rockchip`

when you purged the linux-image-6.1.0-1023-rockchip, it also purged the meta package, which keeps the kernel up to date

Bluesmanuk, 2024-08-29:
I managed to get my NVME install back using the instructions above.

But now I am confused as to where things are supposed to be.

Things are back at 1023 but there is mention of 1025.

But if I read the quote above correctly "when you purged the linux-image-6.1.0-1023-rockchip, it also purged the meta package, which keeps the kernel up to date", this would seem to indicate that there is no path forward because if the meta package required to keep the kernel up to date has been purged, then what process is required to keep up with future updates 1015 and beyond?

JanV123, 2024-08-29:
> `sudo apt install linux-image-rockchip`
>
> when you purged the linux-image-6.1.0-1023-rockchip, it also purged the meta package, which keeps the kernel up to date

Got it. Thanks.

> I managed to get my NVME install back using the instructions above.
>
> But now I am confused as to where things are supposed to be.
>
> Things are back at 1023 but there is mention of 1025.
>
> But if I [...]

By doing the steps to recover your device, you revert back to 1023 but it won't update to the latest kernel (currently 1025) unless you do `sudo apt install linux-image-rockchip`

Bluesmanuk, 2024-08-29:
I had run that and it made no difference but after another reboot and running it again, it stuck.

Thanks.

JanV123, 2024-08-29:
> I had run that and it made no difference but after another reboot and running it again, it stuck.
> 
> Thanks.

kernel updates always require a restart to apply

lucianor, 2024-08-30:
And you have to `sudo apt update` before doing that so you get 1025 version - if you get 1024, you will be stuck again.