Joshua-Riek/ubuntu-rockchip #1016: Bug Report: Ubuntu Desktop setup stuck on "Configuring system locales"

marijnbent, 2024-08-28:
### What happened?

I used balenaEtcher to burn the SD card. Setup goes well, till this screen. I have used both the 22.04 desktop image with a configured WiFi, and 24.04 without configuring setting up a WiFi connection. Both times, the result is the same, see image.

[image]

### Kernel version

Latest release, I cannot use the terminal

### SBC model

Radxa Zero 3W

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Joshua-Riek, 2024-08-28:
How much ram does the device have? Ubuntu desktop requires at least 4GB.

Gi-0n, 2024-08-28:
Your board has heat sinks, it is possible that the heat creates freezing

marijnbent, 2024-08-30:
Thank you anyway @Joshua-Riek, for all the effort!

@Gi-0n I don't have heatsinks installed, those are add-ons right?

closed 2025-02-10 by github-actions[bot]