Joshua-Riek/ubuntu-rockchip #1020: Bug Report: Rock 5b+ Ubuntu don't see Bluetooth devices.

tpnthr, 2024-08-30:
### What happened?

Rock 5b+ ubuntu don't see Bluetooth devices.
Bluetooth device is discovered
Services are running
Blueman is installed, but still no devices shown nearby.
Can anyone help me with that?

### Kernel version

6.1.0-1025-rockchip

### SBC model

rock 5b+

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
Bluetooth Service Status:
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
     Active: active (running) since Wed 2024-08-28 23:45:53 CEST; 1 day 18h ago
       Docs: man:bluetoothd(8)
   Main PID: 785 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 28132)
     Memory: 3.3M (peak: 3.8M)
        CPU: 84ms
     CGroup: /system.slice/bluetooth.service
             └─785 /usr/libexec/bluetooth/bluetoothd

Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/faststream
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/faststream_duplex
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink/opus_05
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/opus_05
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Aug 28 23:45:55 radxa-desktop bluetoothd[785]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Bluetooth Devices List:
Controller A8:41:F4:BC:48:77 radxa-desktop [default]
Bluetooth Adapter Status:
Controller A8:41:F4:BC:48:77 (public)
	Manufacturer: 0x005d (93)
	Version: 0x0b (11)
	Name: radxa-desktop
	Alias: radxa-desktop
	Class: 0x007c0000 (8126464)
	Powered: yes
	Discoverable: no
	DiscoverableTimeout: 0x00000000 (0)
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0548
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0a (10)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
Scan Results:
SetDiscoveryFilter success
Bluetooth Adapter Info:
Devices:
	hci0	A8:41:F4:BC:48:77
hci0:	Type: Primary  Bus: USB
	BD Address: A8:41:F4:BC:48:77  ACL MTU: 1021:6  SCO MTU: 255:12
	UP RUNNING 
	RX bytes:988 acl:0 sco:0 events:92 errors:0
	TX bytes:5593 acl:0 sco:0 commands:93 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'radxa-desktop'
	Class: 0x7c0000
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Miscellaneous, 
	HCI Version: 5.2 (0xb)  Revision: 0xb
	LMP Version: 5.2 (0xb)  Subversion: 0x8852
	Manufacturer: Realtek Semiconductor Corporation (93)

Bluetooth Logs from dmesg:
[   13.464161] Bluetooth: Core ver 2.22
[   13.464176] NET: Registered PF_BLUETOOTH protocol family
[   13.464179] Bluetooth: HCI device and connection manager initialized
[   13.464185] Bluetooth: HCI socket layer initialized
[   13.464189] Bluetooth: L2CAP socket layer initialized
[   13.464201] Bluetooth: SCO socket layer initialized
[   14.688067] usb 3-1.3: Product: Bluetooth Radio
[   16.896711] Bluetooth: hci0: Failed to read codec capabilities (-22)
[   17.721979] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.721995] Bluetooth: BNEP filters: protocol multicast
[   17.722011] Bluetooth: BNEP socket layer initialized
[   17.724852] Bluetooth: MGMT ver 1.22
[   20.274338] Bluetooth: RFCOMM TTY layer initialized
[   20.274381] Bluetooth: RFCOMM socket layer initialized
[   20.274411] Bluetooth: RFCOMM ver 1.11
```

hanvinke, 2024-08-30:
@tpnthr 

Bluetooth seems to work here. I needed also to activate the search for BT devices on my phone.

> **hcitool dev**

Devices:
	hci0	A8:41:F4:A4:03:9D

>**hcitool -i hci0 scan**
             
Scanning ...

And I can pair the discovered device : 
![Screenshot from 2024-08-30 19-37-36](https://github.com/user-attachments/assets/dd41fdb2-cc8d-4499-9be8-2584d58fc19

[https://simpleit.rocks/linux/shell/connect-to-bluetooth-from-cli/](url)

hanvinke, 2024-08-30:
Maybe to some extend relevant because I did some tweaks as well:

`sudo nano /etc/modprobe.d/blacklist.conf`

blacklist pgdrv
blacklist btusb
blacklist btrtl
blacklist btbcm
blacklist btintel

[https://github.com/Joshua-Riek/ubuntu-rockchip/issues/286](url)

`journalctl -u bluetooth.service `

 Aug 30 19:33:16 han-desktop bluetoothd[775]: /org/bluez/hci0/dev_3C_13_5A_21_43_E9/fd0: fd(42) ready

To get rid of : **bluetoothd: sap-server: Operation not permitted** error :

[https://raspberrypi.stackexchange.com/questions/40839/sap-error-on-bluetooth-service-status](url)

tpnthr, 2024-08-30:
Done everything above :(

`root@radxa-desktop:/home/radxa# journalctl -u bluetooth.service
Aug 19 13:37:27 radxa-desktop systemd[1]: Starting bluetooth.service - Bluetooth service...
Aug 19 13:37:27 radxa-desktop (uetoothd)[768]: bluetooth.service: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755>
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: Bluetooth daemon 5.72
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: Starting SDP server
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: src/plugin.c:plugin_init() System does not support csip plugin
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: src/plugin.c:plugin_init() System does not support micp plugin
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: src/plugin.c:plugin_init() System does not support vcp plugin
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: src/plugin.c:plugin_init() System does not support mcp plugin
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: src/plugin.c:plugin_init() System does not support bass plugin
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: src/plugin.c:plugin_init() System does not support bap plugin
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: Bluetooth management interface 1.22 initialized
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: Battery Provider Manager created
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Aug 19 13:37:27 radxa-desktop bluetoothd[768]: sap-server: Operation not permitted (1)
Aug 19 13:37:27 radxa-desktop systemd[1]: Started bluetooth.service - Bluetooth service.
Aug 19 13:37:29 radxa-desktop bluetoothd[768]: Endpoint registered: sender=:1.39 path=/MediaEndpoint/A2DPSource/ldac
Aug 19 13:37:29 radxa-desktop bluetoothd[768]: Endpoint registered: sender=:1.39 path=/MediaEndpoint/A2DPSink/aptx_hd
Aug 19 13:37:29 radxa-desktop bluetoothd[768]: Endpoint registered: sender=:1.39 path=/MediaEndpoint/A2DPSource/aptx_hd
Aug 19 13:37:29 radxa-desktop bluetoothd[768]: Endpoint registered: sender=:1.39 path=/MediaEndpoint/A2DPSink/aptx
Aug 19 13:37:29 radxa-desktop bluetoothd[768]: Endpoint registered: sender=:1.39 path=/MediaEndpoint/A2DPSource/aptx
Aug 19 13:37:29 radxa-desktop bluetoothd[768]: Endpoint registered: sender=:1.39 path=/MediaEndpoint/A2DPSink/sbc
Aug 19 13:37:29 radxa-desktop bluetoothd[768]: Endpoint registered: sender=:1.39 path=/MediaEndpoint/A2DPSource/sbc
`

tpnthr, 2024-08-30:
and now:

`radxa@radxa-desktop:~$ sudo systemctl status bluetooth
[sudo] password for radxa: 
× bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset>
    Drop-In: /etc/systemd/system/bluetooth.service.d
             └─01-disable-sap-plugin.conf
     Active: failed (Result: exit-code) since Fri 2024-08-30 21:23:42 CEST; 21s>
       Docs: man:bluetoothd(8)
    Process: 1937 ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap (code=>
   Main PID: 1937 (code=exited, status=203/EXEC)
        CPU: 21ms

Aug 30 21:23:42 radxa-desktop systemd[1]: bluetooth.service: Scheduled restart >
Aug 30 21:23:42 radxa-desktop systemd[1]: bluetooth.service: Start request repe>
Aug 30 21:23:42 radxa-desktop systemd[1]: bluetooth.service: Failed with result>
Aug 30 21:23:42 radxa-desktop systemd[1]: Failed to start bluetooth.service - B>
`

tpnthr, 2024-08-30:
[https://raspberrypi.stackexchange.com/questions/40839/sap-error-on-bluetooth-service-status](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/url) 
This instruction messed everything :)

tpnthr, 2024-08-30:
`radxa@radxa-desktop:/$ sudo systemctl status bluetooth
Warning: The unit file, source configuration file or drop-ins of bluetooth.serv>
× bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset>
    Drop-In: /etc/systemd/system/bluetooth.service.d
             └─01-disable-sap-plugin.conf
     Active: failed (Result: exit-code) since Fri 2024-08-30 21:26:31 CEST; 5mi>
       Docs: man:bluetoothd(8)
    Process: 4396 ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap (code=>
   Main PID: 4396 (code=exited, status=203/EXEC)
        CPU: 21ms

Aug 30 21:26:31 radxa-desktop systemd[1]: bluetooth.service: Scheduled restart >
Aug 30 21:26:31 radxa-desktop systemd[1]: bluetooth.service: Start request repe>
Aug 30 21:26:31 radxa-desktop systemd[1]: bluetooth.service: Failed with result>
Aug 30 21:26:31 radxa-desktop systemd[1]: Failed to start bluetooth.service - B>
lines 1-15/15 (END)
radxa@radxa-desktop:/$ sudo systemctl restart bluetooth.service
Warning: The unit file, source configuration file or drop-ins of bluetooth.service changed on disk. Run 'systemctl daemon-reload' to reload units.
Job for bluetooth.service failed because the control process exited with error code.
See "systemctl status bluetooth.service" and "journalctl -xeu bluetooth.service" for details.
radxa@radxa-desktop:/$ systemctl status bluetooth.service
Warning: The unit file, source configuration file or drop-ins of bluetooth.serv>
× bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset>
    Drop-In: /etc/systemd/system/bluetooth.service.d
             └─01-disable-sap-plugin.conf
     Active: failed (Result: exit-code) since Fri 2024-08-30 21:32:57 CEST; 18s>
       Docs: man:bluetoothd(8)
    Process: 4833 ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap (code=>
   Main PID: 4833 (code=exited, status=203/EXEC)
        CPU: 22ms

Aug 30 21:32:57 radxa-desktop systemd[1]: bluetooth.service: Scheduled restart >
Aug 30 21:32:57 radxa-desktop systemd[1]: bluetooth.service: Start request repe>
Aug 30 21:32:57 radxa-desktop systemd[1]: bluetooth.service: Failed with result>
Aug 30 21:32:57 radxa-desktop systemd[1]: Failed to start bluetooth.service - B>

radxa@radxa-desktop:/$ journalctl -xeu bluetooth.service
░░ The job identifier is 5600 and the job result is failed.
Aug 30 21:32:57 radxa-desktop systemd[1]: bluetooth.service: Scheduled restart >
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ Automatic restarting of the unit bluetooth.service has been scheduled, as th>
░░ the configured Restart= setting for the unit.
Aug 30 21:32:57 radxa-desktop systemd[1]: bluetooth.service: Start request repe>
Aug 30 21:32:57 radxa-desktop systemd[1]: bluetooth.service: Failed with result>
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit bluetooth.service has entered the 'failed' state with result 'exit->
Aug 30 21:32:57 radxa-desktop systemd[1]: Failed to start bluetooth.service - B>
░░ Subject: A start job for unit bluetooth.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit bluetooth.service has finished with a failure.
░░ 
░░ The job identifier is 5687 and the job result is failed.
`

hanvinke, 2024-08-30:
@tpnthr 

That is unfortunate!  I suppose that you did a 'systemctl daemon-reload' to reload bluetoothd daemon? BTW I only did the first answer with _Add "--noplugin=sap" at the end of the "ExecStart" Line_ .

Joshua-Riek, 2024-08-30:
AIC8800 Bluetooth is kind of fucked in general, I would look at aic8800-bluetooth.service or the following code.
```bash
#!/bin/bash

rfkill unblock all
/usr/bin/hciattach -s 1500000 /dev/ttyS1 any 1500000 flow nosleep
sleep 2

while read -r; do
    echo "bt_test > $REPLY"
    case "$(tr -d '\r' <<< "$REPLY")" in
        "hci recv thread ready (nil)")
            echo "Device reset successfully."
            exit 0
            ;;
        "dev_open fail")
            echo "Unable to open /dev/ttyS1. Is Bluetooth already up?"
            exit 1
            ;;
    esac
done < <(timeout 1 bt_test -s uart 1500000 "/dev/ttyS1")

echo "Command timed out."
exit 2
```

hanvinke, 2024-08-30:
Mmmm.. maybe I did some other stuff too?!  I also added experimental support to Ubuntu.

> To enable Bluez experimental features in Ubuntu:

-Open /etc/bluetooth/main.conf as root
-Find the section titled ‘General’ (near the top)
-On a new line add Experimental = true
-Save the file (and hit save again to make sure)
-Reboot or run systemctl restart bluetooth

Don’t skip the last step – if you do, nothing will change.

On my system current state of bluetooth server:

` systemctl status bluetooth`

_● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
     Active: active (running) since Fri 2024-08-30 22:09:17 CEST; 58min ago
       Docs: man:bluetoothd(8)
   Main PID: 798 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 28132)
     Memory: 3.6M (peak: 4.1M)
        CPU: 88ms
     CGroup: /system.slice/bluetooth.service
             └─798 /usr/libexec/bluetooth/bluetoothd --noplugin=sap_

tpnthr, 2024-08-30:
Now it shows me 
`radxa@radxa-desktop:~$ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
     Active: active (running) since Fri 2024-08-30 23:31:50 CEST; 38s ago
       Docs: man:bluetoothd(8)
   Main PID: 3336 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 28132)
     Memory: 756.0K (peak: 1.2M)
        CPU: 51ms
     CGroup: /system.slice/bluetooth.service
             └─3336 /usr/libexec/bluetooth/bluetoothd --noplugin=sap

Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: Starting SDP server
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: Excluding (cli) sap
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: src/plugin.c:plugin_init() System does not support csip plugin
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: src/plugin.c:plugin_init() System does not support micp plugin
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: src/plugin.c:plugin_init() System does not support vcp plugin
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: src/plugin.c:plugin_init() System does not support mcp plugin
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: src/plugin.c:plugin_init() System does not support bass plugin
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: src/plugin.c:plugin_init() System does not support bap plugin
Aug 30 23:31:50 radxa-desktop bluetoothd[3336]: Bluetooth management interface 1.22 initialized
`

tpnthr, 2024-08-30:
interface shows me that
[image]

tpnthr, 2024-08-30:
`radxa@radxa-desktop:~$ dmesg | grep -i bluetooth
[   13.529594] Bluetooth: Core ver 2.22
[   13.529609] NET: Registered PF_BLUETOOTH protocol family
[   13.529612] Bluetooth: HCI device and connection manager initialized
[   13.529618] Bluetooth: HCI socket layer initialized
[   13.529622] Bluetooth: L2CAP socket layer initialized
[   13.529634] Bluetooth: SCO socket layer initialized
[   15.320262] usb 3-1.3: Product: Bluetooth Radio
[   25.981332] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   25.981348] Bluetooth: BNEP filters: protocol multicast
[   25.981364] Bluetooth: BNEP socket layer initialized
[  514.977207] Bluetooth: hci0: Failed to read codec capabilities (-22)
[  514.980504] Bluetooth: MGMT ver 1.22
[  514.988695] Bluetooth: RFCOMM TTY layer initialized
[  514.988715] Bluetooth: RFCOMM socket layer initialized
[  514.988734] Bluetooth: RFCOMM ver 1.11
[  541.249398] Bluetooth: hci0: urb 0000000066412dbb failed to resubmit (2)
[  541.563207] Bluetooth: hci0: Failed to read codec capabilities (-22)
[  541.566641] Bluetooth: MGMT ver 1.22
radxa@radxa-desktop:~$ 
`

hanvinke, 2024-08-31:
@tpnthr 

Looks good actually. 

My journalctl and lsmod:

[journalctl.txt](https://github.com/user-attachments/files/16824086/journalctl.txt)

[lsmod.txt](https://github.com/user-attachments/files/16824093/lsmod.txt)

[image]

tpnthr, 2024-09-11:
sudo modprobe rtk_btusb && dmesg | grep rtk - worked like magic!!! Thank you

closed 2025-02-10 by github-actions[bot]