Joshua-Riek/ubuntu-rockchip #1022: Feature Request: How to Integrate kenel 6.10.6-current-rockchip-rk3588 Kernel from Armbian_24.8.1  into Ubuntu Joshua 24.04 v2.3.1  on Orange Pi 5 Pro?

dulimov, 2024-08-31:
### Feature Description

Hello everyone,

I have an Orange Pi 5 Pro running the latest Ubuntu Joshua distribution. I'm interested in integrating the img-6.10.6-current-rockchip-rk3588 kernel, which is provided by Armbian for the same board, into my current Ubuntu setup.

Questions:

What are the necessary steps to properly integrate this kernel from Armbian into my Ubuntu Joshua system?
Are there any potential compatibility issues that I should be aware of, and will the system work seamlessly after this kernel is installed?
Additional Context:

The kernel version I’m trying to install is designed for the same hardware (Orange Pi 5 Pro) but comes from the Armbian distribution.
This kernel (6.10.6-current-rockchip-rk3588) was built on August 27, 2024.
The goal is to ensure that all hardware components work correctly after the kernel switch, including peripherals and built-in components.
Any advice or guidance would be greatly appreciated!

Thank you!

Joshua-Riek, 2024-08-31:
I don't think you are aware of the full kernel situation; mainline Linux does not fully support RK3588 hardware. Additionally, the Orange Pi 5 Pro is not supported in the 6.10 Armbian kernel. 

I have nightly builds with Ubuntu 24.10 and Linux 6.11-rc1, but it only supports select hardware with limited functionality. 
See: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/nightly.yml

dulimov, 2024-08-31:
hank you for the information! I monitor updates daily, hoping my board will get better support. However, I noticed that on this link, my board is listed with the claimed support for the new 6.10 kernel. Could you check if this is an error?

https://uk.mirrors.naho.moe/armbian-dl/orangepi5pro/archive/

Joshua-Riek, 2024-08-31:
Interesting, I guess someone added a device tree for the board.

closed 2024-09-07 by Joshua-Riek