Joshua-Riek/ubuntu-rockchip #1024: Feature Request: RK3582 support

Qengineering, 2024-09-01:
### Feature Description

It would be great if your awesome OS would also support the RK3582 which can be found on the Rock5C-lite board among others

Joshua-Riek, 2024-09-01:
It's impossible to add support for an entire SOC, but the Rock 5C Lite is already supported, please use the Rock 5C image.

closed 2024-09-01 by Joshua-Riek

Qengineering, 2024-09-01:
It is not about the entire SOC, an Ubuntu OS, like the one for the Rock 5C, would be great!

At the moment the current Rock5C version found on the release download page doesn't work.

After the initialization steps Ubuntu 22.04 [release 2024-08-16](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.3.1/ubuntu-22.04-preinstalled-desktop-arm64-rock-5c.img.xz) reboots and the Rock5C-Lite errors with:
 `fail to start casper-m5check Verifiy Live ISO checksums`.

The Ubuntu 24.04 [release 2024-08-16](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.3.1/ubuntu-24.04-preinstalled-desktop-arm64-rock-5c.img.xz) shows in a glimpse the Ubuntu 24.04 screen, jumps to command line messages and stocks with only a blinking cursor in the top. No initialization whatsoever.

Joshua-Riek, 2024-09-01:
Oh, sounds like there is a bug then, not that its not supported.

Qengineering, 2024-09-01:
Indeed. A bug report?